facebookresearch/habitat-lab #230: Semantics: Instance level -> 40-category level

lynl7130, 2019-10-18:
Hi,

I used to think d3_40_colors_rgb function as a method to translate the instance level annotation to 40-category level annotation. 
However, In Matterport3D, scene_id = 2azQ1b91cZZ, I see the following weird output from my semantic sensor:

[image]

This is an output with d3_40_colors_rgb applied. You can see that every chair around the table has a different color, which does not make sense.

I followed exactly what you demonstrated in your tutorial, and I plotted out all of the object.catetory.index(), looks like all chairs are labeled '3' correctly.  

So how could I map instance-level labels to category-level labels?

Thanks in advance!

erikwijmans, 2019-10-18:
The `d3_40_colors_rgb` is just for colorizing semantic or instance IDs.  The objects themselves can be used to creating the mapping
```python
object_id_to_class_id = {int(obj.id.split("_")[-1]): obj.category.index() for obj in scene.objects}
```

lynl7130, 2019-11-04:
Problem solved, thank you!

closed 2019-11-04 by lynl7130