facebookresearch/habitat-lab #232: Instruction on sensor usages

chenwydj, 2019-10-18:
## ❓ Questions and Help

I wonder if a detailed instruction on the supported sensors would be possible?

For some of the sensors I can only guess their meanings. For example, I can see **PointGoalWithGPSCompassSensor** provides a dynamic two-dim vector. Is this the target coordination or just a distance and the orientation?

Thank you!

mathfac, 2019-10-22:
Hi @chenwydj 
There is a doc string explaining different goal formats:
```
Sensor that integrates PointGoals observations (which are used PointGoal Navigation) and GPS+Compass.
    For the agent in simulator the forward direction is along negative-z.
    In polar coordinate format the angle returned is azimuth to the goal.
    Args:
        sim: reference to the simulator for calculating task observations.
        config: config for the PointGoal sensor. Can contain field for
            GOAL_FORMAT which can be used to specify the format in which
            the pointgoal is specified. Current options for goal format are
            cartesian and polar.
            Also contains a DIMENSIONALITY field which specifies the number
            of dimensions used to specify the goal, must be in [2, 3]
    Attributes:
        _goal_format: format for specifying the goal which can be done
            in cartesian or polar coordinates.
        _dimensionality: number of dimensions used to specify the goal
```
`Cartesian coordinates` are target coordination, `polar coordinates` are a distance and the orientation.

chenwydj, 2019-10-24:
Thank you mathfac!

Am I correct that the current PointGoalWithGPSCompassSensor is dynamic (i.e. the sensor will dynamically update its value according to agent's location, although the target position is fixed)

erikwijmans, 2019-10-24:
Yes, the `PointGoalWithGPSCompassSensor` sensor returns the goal relative to agents current location (as opposed to the `PointGoalSensor`, which always returns the goal relative to the starting location).  For both sensor, the goal point is always static wrt. the world coordinate frame (i.e. the mesh's coordinates).

chenwydj, 2019-10-24:
Thank you Erik!

chenwydj, 2019-11-19:
@mathfac Could you give some introduction (physical meanings, value range, etc.) about these following sensors:
'HEADING_SENSOR', 'COMPASS_SENSOR', 'GPS_SENSOR', 'PROXIMITY_SENSOR'

Thank you very much!

erikwijmans, 2019-11-19:
We need to do a better job exposing these in docs, but these sensors do have reasonably informative doc strings: https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav.py#L258

chenwydj, 2019-11-19:
Thank you Erik!

For the ProximitySensor:
1) Is the obstacle measured observed the current observation, or just the closest one around the agent?
2) How do I understand the range of the proximity value? By running the a job in matterport3d I found this value ranges from 0 ~ 2. Is this value measured in the same unit as the GPS sensor?

Thanks!

erikwijmans, 2019-11-19:
It's always the closest obstacle to the agent's current location.  It does not take into account where the agent is looking.

As with anything in habitat, the units of any value will depend on the dataset that you are loading as everything will be in the units of that dataset, for MP3D/Gibson/Replica, it's all meters.

mathfac, 2020-02-06:
Closing as all questions should be answered.

closed 2020-02-06 by mathfac