facebookresearch/habitat-lab #239: What is point of `prev_actions`?

chenwydj, 2019-10-30:
## ❓ Questions and Help

The `prev_actions` is defined in rollouts [here](https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/common/rollout_storage.py#L53).

In `ppo_trainer.py`, although `prev_actions` is passed into `self.actor_critic.act()`, `self.actor_critic.get_value()`, and `self.actor_critic.evaluate_actions()`, it was not used in `self.net` ([line here](https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/rl/ppo/policy.py#L162))

May I ask the motivation of `prev_actions`, and any potential place to use it?

Thank you!

erikwijmans, 2019-10-30:
It's there in case people need it.  It can be beneficial to explicitly give the policy the previous action taken as input, (i.e. modeling `p(a_t | s_t)` as `p(a_t | o_t, a_{t-1}, h_t)` instead of `p(a_t | o_t, h_t)`).   The previous action is also needed for self-supervision objectives like next observation prediction.

chenwydj, 2019-10-30:
Thank you Erik!

closed 2019-10-30 by chenwydj