facebookresearch/habitat-lab #240: How to reduce the cpu usage?

chenwydj, 2019-10-30:
## ❓ Questions and Help

I found the habitat consumes a quite high cpu usage (i.e. number of processes occupied).

Is there any way to adjust the number of processes (workers) used by habitat?

Thank you!

kowshikthopalli, 2019-10-30:
Are you talking about when using the baselines code. One can change the number of processes in th e config file here 
https://github.com/facebookresearch/habitat-api/blob/5fd780796fc919680b2783851db7245a3a8db3b6/habitat_baselines/config/pointnav/ppo_pointnav.yaml#L11

chenwydj, 2019-10-31:
> Are you talking about when using the baselines code? One can change the number of processes in th e config file here
> https://github.com/facebookresearch/habitat-api/blob/5fd780796fc919680b2783851db7245a3a8db3b6/habitat_baselines/config/pointnav/ppo_pointnav.yaml#L11

Thank you for your reply!

I already set `NUM_PROCESSES` as 1.

Does that mean I only have one habitat process that rendering the observations?

mathfac, 2019-10-31:
Hi @chenwydj,
> Does that mean I only have one habitat process that rendering the observations?

That's right.

To better answer your question can you provide output of the `collect_env` script and system info about number and spec of CPU/GPU of your machine:
```
wget https://raw.githubusercontent.com/facebookresearch/habitat-sim/master/habitat_sim/utils/collect_env.py
python3 collect_env.py
```

Is occupation always 100% or only during particular steps? You can try to comment out training or backprop step to check what exactly brings CPU load: simulator or training.

chenwydj, 2019-11-19:
After debugging I found it may be my pytorch code that consumes high cpu threds. I solved it via `torch.set_num_threads(1)`.

Thank you!

closed 2019-11-19 by chenwydj