facebookresearch/habitat-lab #241: Episode iterator does not respect seed

Danmou, 2019-11-01:
## 🐛 Bug

There are currently two ways to set the seed used for random generators in Habitat:
* The `SEED` option in the config
* The `env.seed(...)` function

Neither of these affect the episode order when shuffle is enabled. As a workaround, setting `random.seed` and `np.random.seed` before calling the constructor works, but this doesn't seem very elegant. It's especially annoying when using VectorEnv, because you have to set the seed inside the `make_env_fn` for it to have an effect in the created subprocess.

rosanom, 2020-04-16:
Does anybody know if the bug was fixed? I generated navigation episodes using @erikwijmans 's script [create_pointnav_dataset.py](https://gist.github.com/erikwijmans/1c596183fb8de1e9afd80088b6b5c115) but repeating the generation process I get the same episodes. Neither the `SEED` of the config file nor the `env.seed(...)` worked. Any solutions?

dhruvbatra, 2020-04-16:
CC: @abhiskk

naokiyokoyama, 2020-09-10:
@mathfac it seems that this is still a bug, when I try to run
`python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type eval`
to test my weights, despite the fact that I keep using the same .yaml file without changing its "SEED" between runs, the order in which episodes are chosen for evaluation is not consistent. Also, the agent's random behavior is not consistent between runs either, when the DDPPO policy is set to deterministic=False.