facebookresearch/habitat-lab #244: How to generate legal random start point on the floor

lynl7130, 2019-11-04:
Hi, 

While using `env._sim.sample_navigable_point()` as random point sampling method with MP3d, I sometimes find the agent will land on objects like sofa or bed instead of on the ground.

How could I get of these situations and force the agent to land on floor?

PS. Does habitat allow agents to pass through object?

Thanks in advance!

mathfac, 2019-11-04:
Hi @lynl7130,

You can use MP3D level annotations and based on bounding box coordinates filter randomly generated points. To learn how to access MP3D level annotations check [relevant tutorial](https://aihabitat.org/docs/habitat-api/habitat-api-demo.html#scene-semantic-annotations).

Alternatively, you can generate reasonable amount of points from same floor and save min/avg `y` coordinate (height). Then use height filtering for generating points similar how it's done for [the PointNav episode generation](https://github.com/facebookresearch/habitat-api/blob/master/habitat/datasets/pointnav/pointnav_generator.py#L39).

lynl7130, 2019-11-05:
Got it. Thank you!

closed 2019-11-05 by lynl7130