facebookresearch/habitat-lab #246: How to write NavigationEpisode to json.gz file?

stbook, 2019-11-05:
## ❓ Questions and Help
I was trying to build my own training episodes by generating start positions and goal positions and NavigationEpisodes.
But when I was writing the built NavigationEpisode into .json.gz file, I get this error
'TypeError: Object of type 'NavigationEpisode' is not JSON serializable'

I couldn't fix it. I was wondering how the scene_name.json.gz file were created?

mathfac, 2019-11-05:
@stbook, are you sure you use `dataset.to_json()` method that employs [Dataset's custom JSON Encoder](https://github.com/facebookresearch/habitat-api/blob/master/habitat/core/dataset.py#L113)?

stbook, 2019-11-05:
Thanks for your quick response. 

I changed `dataset.to_json()` a bit by converting `numpy.float32` to `float` and it works now.

I have a follow-up question. After we created the dataset should we use the following line to write it directly to disk, because I run into some errors here. I was wondering if there is a better way or an API to use.
`with gzip.GzipFile('../data/datasets/pointnav/mp3d/v1/my/temp.json.gz', 'w') as fout:
    fout.write(dataset)`

mathfac, 2019-11-05:
That's important to do "utf-8" encoding:
```
    json_str = str(dataset.to_json())
    with gzip.GzipFile(output_path + ".json.gz", 'wb') as f:
        f.write(json_str.encode("utf-8"))
```

stbook, 2019-11-05:
It works like a charm. Thanks!

closed 2019-11-05 by stbook