facebookresearch/habitat-lab #247: agent state to 2D pose

stbook, 2019-11-07:
## ❓ Questions and Help
I was trying to convert agent state into a 2D pose tuple (x, y, theta). I followed some existent code, but the result doesn't seem to be correct.

Here is the code.

```
position = sim.get_agent_state().position
rot = sim.get_agent_state().rotation
pose = [0.0, 0.0, 0.0]
pose[0] = position[0] # for x
pose[1] = position[2] # for y

direction_vector = np.array([0, 0, -1])
heading_vector = quaternion_rotate_vector(rot.inverse(), direction_vector)
pose[2] = cartesian_to_polar(-heading_vector[2], heading_vector[0])[1] # for theta

```

erikwijmans, 2019-11-07:
Can you give more information about why "the result doesn't seem to be correct"?  This is the 2D pose of the agent in some coordinate system, but maybe not the coordinate system you are expecting.

stbook, 2019-11-07:
I want the agent to be in 2d world coordinate system that x,y-coordinates achieve minimum value at left-bottom corner.
[image]

Using the current code I give, the pose is not consistent when the agent moves in some direction.

erikwijmans, 2019-11-07:
You'll need to correct the angle like here: https://github.com/facebookresearch/habitat-api/issues/109#issuecomment-499668103

stbook, 2019-11-07:
I've computed the angle like that before by adding `pi/2` to the negative angle but it doesn't seem to be right.
For example, when the agent turns left by `pi/6`, the pose `(x, y, theta)` changes from `(7.35, -4.61, 0.74)` to `(7.35, -4.61, 0.21)`, which should be `(7.35, -4.61, 0.74 + pi/6) = (7.35, -4.61, 1.25)` if the computation is right.
When the agent moves forward by `0.3`, the agent moves from `(6.76, -4.74, 0.21)` to `(6.47, -4.80, 0.21)`, which is not right as the correct next pose should be `(6.76 + 0.3*cos(0.21), -4.74 + 0.3*sin(0.21), 0.21) = (7.06, -4.74, 0.21)` if the computation is right.

Sorry to bother you so much.

mathfac, 2019-11-07:
@stbook, does `HEADING_SENSOR` work for you? You can use `theta` from  `observations["heading"]`.
Otherwise, that would be cleaner if you can provide test that shouldn't fail but fails in form like this: 
```
def test_heading_vector():
    config=habitat.get_config("./configs/tasks/pointnav.yaml")
    env = habitat.Env(config)
    observations = env.reset()
    state_start = env.sim.get_agent_state()
    env.step("TURN_RIGHT")
    env.step("TURN_RIGHT")
    state_final = env.sim.get_agent_state()
    ... <your code calculating angles> ...
    assert actual_angle == calculated_angle
    env.close()
```
That will make easy to reproduce the setup, thanks.

erikwijmans, 2019-11-07:
@stbook -- Looks like the link I pointed you to is for a different coordinate system all-together :) 

It seems like you want something that is like the TopDownMap measure.  The way we get the angle for that is here: https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav.py#L667

erikwijmans, 2019-11-07:
The subtraction of pi/2 deals with the flip of x and y that happens when you assign x to x and z to y.  In the agents coordinate system, z corresponds to theta=0, not theta=pi/2.

stbook, 2019-11-08:
Thanks a lot.

closed 2019-11-09 by stbook