facebookresearch/habitat-lab #253: Check failed: std::getenv("DISPLAY") != nullptr DISPLAY not detected

BryanBo-Cao, 2019-11-16:
## ❓ Questions and Help
I tried to run
```
python examples/shortest_path_follower_example.py
```
but got this error
```
2019-11-15 22:08:59,608 Initializing dataset PointNav-v1
2019-11-15 22:08:59,853 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
F1115 22:08:59.854769  6454 WindowlessContext.cpp:235] Check failed: std::getenv("DISPLAY") != nullptr DISPLAY not detected. For headless systems, compile with --headless for EGL support
*** Check failure stack trace: ***
Aborted (core dumped)
```
Any help would be appreciated!

erikwijmans, 2019-11-16:
Is this a system with an attached display?  Or are you on a remote server?

BryanBo-Cao, 2019-11-16:
Hi @erikwijmans, thanks for your reply!

By "an attached display?" do you mean a physical monitor? I am afraid not.
I was running the code on a remote server:
```
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.3 LTS
```
But I have not installed X server yet.

erikwijmans, 2019-11-16:
Yeah, generally you need an attached physical monitor to have a window server (which is what non-headless mode requires).  If the system has an Nvidia GPU, you can build habitat-sim in headless mode with by adding `--headless` to the install command.  

If not, you can try to get Xvfb to work, but Xvfb has been flakey in my experience.

BryanBo-Cao, 2019-11-16:
Thanks @erikwijmans . I rebuilt ```habitat-sim``` by ```--headless``` according to https://github.com/facebookresearch/habitat-sim. Now I can run the example code in ```habitat-api```.

closed 2019-11-16 by BryanBo-Cao