facebookresearch/habitat-lab #258: How to limit the agent on single-floor rooms?

chenwydj, 2019-11-26:
## ❓ Questions and Help
I found current MP3D dataset contains rooms with multi-floor rooms.

Is there any way to filter out multi-floor rooms, only keeping single-floor ones?

Thank you very much!

chenwydj, 2019-12-06:
Could anyone lend some help? Thank you!

erikwijmans, 2019-12-06:
In the case of MP3D, filtering to just single floor houses will likely leave very few remaining.  If you want to do that, you can use the semantic annotations to do that: https://github.com/facebookresearch/habitat-api/blob/master/habitat/sims/habitat_simulator/habitat_simulator.py#L328

Alternatively, you can look into re-generating the navmeshes (the things that do collision checking/stair climbing/etc) with a lower agent climb angle (https://github.com/facebookresearch/habitat-sim/blob/master/src/esp/nav/PathFinder.h#L66).  Either using https://github.com/facebookresearch/habitat-sim/pull/333 or https://github.com/facebookresearch/habitat-sim/blob/master/src/utils/datatool/datatool.cpp

chenwydj, 2019-12-07:
Thank you very much Erik!

mathfac, 2020-01-10:
Closing as answered.

closed 2020-01-10 by mathfac