facebookresearch/habitat-lab #260: Is it appropriate to call worker_conn.close() just after ps.start() without ps.join() in _spawn_workers?

xiaotaw, 2019-11-28:
## ❓ Questions and Help
Sorry, I did not run any test code because had not found any, and just wondering if is it safe to call `worker_conn.close()` just after `ps.start()` without `ps.join()`? 

It seems that `worker_conn.recv` and `worker_conn.send` are called continuously in the subprocess.

Thank you! 

https://github.com/facebookresearch/habitat-api/blob/master/habitat/core/vector_env.py#L251

```python
    def _spawn_workers(
        self,
        env_fn_args: Sequence[Tuple],
        make_env_fn: Callable[..., Union[Env, RLEnv]] = _make_env_fn,
    ) -> Tuple[List[Callable[[], Any]], List[Callable[[Any], None]]]:
        parent_connections, worker_connections = zip(
            *[self._mp_ctx.Pipe(duplex=True) for _ in range(self._num_envs)]
        )
        self._workers = []
        for worker_conn, parent_conn, env_args in zip(
            worker_connections, parent_connections, env_fn_args
        ):
            ps = self._mp_ctx.Process(
                target=self._worker_env,
                args=(
                    worker_conn.recv,
                    worker_conn.send,
                    make_env_fn,
                    env_args,
                    self._auto_reset_done,
                    worker_conn,
                    parent_conn,
                ),
            )
            self._workers.append(ps)
            ps.daemon = True
            ps.start()
            worker_conn.close()
        return (
            [p.recv for p in parent_connections],
            [p.send for p in parent_connections],
        )
```

erikwijmans, 2019-11-28:
Yes, the parent closing the worker end and the worker closing the parent end immediately is customary with Unix pipes.

xiaotaw, 2019-11-28:
Thank you for the reply. 

Hmm, is it right that showed as follow? (√: opened, x: closed)

| Initiating pipe | parent_end | worker_end |
|-|-|-|
| parent_process | √ | √ |
| worker_process | not exist | not exist |

| Creating worker process | parent_end | worker_end |
|-|-|-|
| parent_process | √ | √ |
| worker_process | √ | √ |

| Working | parent_end | worker_end |
|-|-|-|
| parent_process | √ | x |
| worker_process | x | √ |

erikwijmans, 2019-11-29:
Yeah, that is the flow of things.

xiaotaw, 2019-11-29:
I Get it. Thank you!

closed 2019-11-29 by xiaotaw