facebookresearch/habitat-lab #261: How to specify certain gibson file, also can we extract poses in certain environment?

Dhagash4, 2019-12-02:
My task is to save the RGB and depth images from the scene and also to extract pose.

I am able to save RGB and Depth images but I cant see in the documentation anything about poses can I access them or save them.

Currently I am using pointnav_mp3d.yaml as the conifg file but I am unable to figure out how to provide different scene than the default, I have dowloaded GIbson data and want to specify certain gibson file as my scene. How to do it/

I have tried this:

`config = habitat.get_config(config_paths='../configs/tasks/pointnav_mp3d.yaml')`
`config.defrost()`
`config.DATASET.DATA_PATH = '../data/datasets/pointnav/mp3d/v1/val/val.json.gz'`
`config.DATASET.SCENES_DIR = '../data/scene_datasets/'`
`config.freeze()`
`env = habitat.Env(config=config)`

I am changing path of this pointing to my .glb file but its not opening my .glb file:

`config.DATASET.SCENES_DIR = '../data/scene_datasets/'`

I would be great if someone can provide insight into this issue,

Thanks

erikwijmans, 2019-12-02:
This example is likely a superset of what you are trying to do: https://aihabitat.org/docs/habitat-api/view-transform-warp.html

The easiest way to swap to Gibson is to load the corresponding PointNav dataset, i.e. `config.DATASET.DATA_PATH = '../data/datasets/pointnav/gibson/v1/train/train.json.gz'`

Dhagash4, 2019-12-03:
Thanks for helping it worked but I had one more doubt that I downloaded the gibson with train.json.gz of diffrent environments but there are many more environment in gibson, so is habitat is limeted to that file only or I can choose some other gibson file other than all ready provide.

For example: I want to get the images of akermenviller.glb which is not in the task dataset provided by pointnav Gibson file.

erikwijmans, 2019-12-03:
The majority if scenes in gibson contain major reconstruction artifacts, so we didn't provide PointNav episodes for them.  The function to generate new episodes is here: https://github.com/facebookresearch/habitat-api/blob/master/habitat/datasets/pointnav/pointnav_generator.py

erikwijmans, 2019-12-03:
If you are just looking to generate images from random view points, using the lower-level API with the simulator may be simpler.  This example shows how to initialize some sensors and render with them: https://github.com/facebookresearch/habitat-sim/blob/master/examples/stereo_agent.py

Calling `sim.initialize_agent(0)` will set the agent to a new, random location/orientation.  You can call `sim.get_sensor_observations()` directly to get observations instead of taking a step also.

Dhagash4, 2019-12-04:
Thanks for help, yeah habitat sim is much simpler but my goal is to take images from certain specific view points and ulimately do slam in those environments...

erikwijmans, 2019-12-04:
The docker image/file was locked to specific versions for compatibility with the 2019 habitat challenge.  You can change the commits to new versions.

You can also remove those version checkouts and make the clones 
`git clone --branch stable git@github.com:facebookresearch/habitat-sim.git`
`git clone --branch stable git@github.com:facebookresearch/habitat-api.git`

To grab the latest release

Dhagash4, 2019-12-05:
Thanks you. I cloned the stable branch but it is not working and having error message as
`AttributeError: module 'habitat_sim' has no attribute 'Configuration'` when I am trying to run example.py after installing the api

erikwijmans, 2019-12-05:
Did you also clone and rebuild habitat-sim?

Dhagash4, 2019-12-06:
Yeah I have cloned and rebuild habitat-sim, I have used build.sh to build and also exporting python path but the same error is reappearing

erikwijmans, 2019-12-06:
Using `python setup.py install` and not touching the python path is recommended

Dhagash4, 2019-12-06:
Thank a lot that really heped, just one last question how to specify scene in pointnav_generator.py and will the code return the json file data so that I can use it later, also where to specify certain .glb and where the output will be?

erikwijmans, 2019-12-06:
This is how I would do it: https://gist.github.com/erikwijmans/1c596183fb8de1e9afd80088b6b5c115

Toraudonn, 2020-01-17:
> The majority of scenes in gibson contain major reconstruction artifacts, so we didn't provide PointNav episodes for them. The function to generate new episodes is here: https://github.com/facebookresearch/habitat-api/blob/master/habitat/datasets/pointnav/pointnav_generator.py

Hey, sorry for interrupting, how do major reconstruction artifacts affect PointNav episodes? Will the agent get stuck somewhere or is the observation just not clean?

mathfac, 2020-02-06:
@Toraudonn the generation code will generate the valid episodes, but environment and trajectories would look strange and far from a navigation in real life. Here is an example of such episode:

[image]

Closing the issue, feel free to open if you have questions.

closed 2020-02-06 by mathfac