facebookresearch/habitat-lab #264: Use LOOK_UP and LOOK_DOWN action

chenwydj, 2019-12-08:
## ❓ Questions and Help
When I add
`POSSIBLE_ACTIONS = ["STOP", "MOVE_FORWARD", "TURN_LEFT", "TURN_RIGHT", "LOOK_UP", "LOOK_DOWN"]`
into [line 17 in configuration here](https://github.com/facebookresearch/habitat-api/blob/master/configs/tasks/pointnav.yaml#L17)

and then run the [example here](https://github.com/facebookresearch/habitat-api/blob/71d409ab214a7814a9bd9b7e44fd25f57a0443ba/docs/pages/quickstart.rst)

I got an error
```bash
  File "/home/chenwy/habitat-api/habitat/core/env.py", line 255, in step
    action=action, episode=self.current_episode
  File "/home/chenwy/habitat-api/habitat/core/embodied_task.py", line 346, in step
    self, **action["action_args"], task=self
  File "/home/chenwy/habitat-api/habitat/tasks/nav/nav_task.py", line 766, in step
    return self._sim.step(SimulatorActions.LOOK_UP)
  File "/home/chenwy/habitat-api/habitat/sims/habitat_simulator/habitat_simulator.py", line 240, in step
    sim_obs = self._sim.step(action)
  File "/home/chenwy/habitat-sim/habitat_sim/simulator.py", line 211, in step
    collided = self._default_agent.act(action)
  File "/home/chenwy/habitat-sim/habitat_sim/agent/agent.py", line 154, in act
    ), f"No action {action_id} in action space"
AssertionError: No action 4 in action space
```

I double checked my `env` has the `LOOK_UP` action:
```python
(Pdb) env.action_space
ActionSpace(LOOK_DOWN:<habitat.core.embodied_task.EmptySpace object at 0x7faec51b5358>, LOOK_UP:<habitat.core.embodied_task.EmptySpace object at 0x7faec51b59b0>, MOVE_FORWARD:<habitat.core.embodied_task.EmptySpace object at 0x7faec51b55f8>, STOP:<habitat.core.embodied_task.EmptySpace object at 0x7faec51b53c8>, TURN_LEFT:<habitat.core.embodied_task.EmptySpace object at 0x7faec51b55c0>, TURN_RIGHT:<habitat.core.embodied_task.EmptySpace object at 0x7faec51b56a0>)
(Pdb) habitat.SimulatorActions.LOOK_UP
4
```

May I know how to properly use the `LOOK_UP` and `LOOK_DOWN` action?

Thank you very much!

erikwijmans, 2019-12-08:
The action space (https://github.com/facebookresearch/habitat-api/blob/master/habitat/config/default.py#L176) needs to be swapped to `v1` for those actions.  We should remove that tho as this is no longer needed CC @mathfac

rpartsey, 2022-09-01:
Stale.

closed 2022-09-01 by rpartsey