facebookresearch/habitat-lab #267: Example does not run

smorad, 2019-12-19:
## ❓ Questions and Help
I've run into two issues with running `example/example.py`

1. The dataset name in the `stable` branch is wrong (fixed in `master`, #252), perhaps this should be cherrypicked into `stable`?
2. There is some strange pointnav import bug I cannot track down present in both `stable` and `master`. Perhaps I've installed the api incorrectly?

```
(habitat) smorad@smorad-ThinkPad-T490:~/code/cambridge/habitat-api$ python3 examples/example.py
2019-12-19 13:19:46,441 Initializing dataset PointNav-v1
Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/home/smorad/code/cambridge/habitat-api/habitat/core/env.py", line 74, in __init__
    id_dataset=config.DATASET.TYPE, config=config.DATASET
  File "/home/smorad/code/cambridge/habitat-api/habitat/datasets/registration.py", line 19, in make_dataset
    return _dataset(**kwargs)
  File "/home/smorad/code/cambridge/habitat-api/habitat/datasets/pointnav/__init__.py", line 38, in __init__
    raise pointnav_import_error
  File "/home/smorad/code/cambridge/habitat-api/habitat/datasets/pointnav/__init__.py", line 20, in _try_register_pointnavdatasetv1
    from habitat.datasets.pointnav.pointnav_dataset import (
  File "/home/smorad/code/cambridge/habitat-api/habitat/datasets/pointnav/pointnav_dataset.py", line
15, in <module>
    from habitat.tasks.nav.nav import (
  File "/home/smorad/code/cambridge/habitat-api/habitat/tasks/nav/nav.py", line 29, in <module>
    from habitat.sims.habitat_simulator.actions import HabitatSimActions
  File "/home/smorad/code/cambridge/habitat-api/habitat/sims/habitat_simulator/actions.py", line 12,
in <module>
    import habitat_sim
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.3-py3.6-linux-x86_64.egg/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.3-py3.6-linux-x86_64.egg/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ImportError: generic_type: type "MapStringString" is already registered!
```

smorad, 2019-12-19:
2. was an issue with habitat-sim and magnum-bindings. See https://github.com/facebookresearch/habitat-sim/issues/396 for a fix

erikwijmans, 2019-12-20:
Based on your shell prompt, it seems you are in a conda env of virtual env, but the error shows that the files are installed in `/usr/local/lib/python3.6/dist-packages` which is weird/concerning.

What does `which python` return?

mathfac, 2020-01-09:
@smorad is the issue resolved on new master?

smorad, 2020-01-10:
It's hard for me to check. I didn't really understand how to use virtualenvs, so I polluted my base system during my install and debugging. Maybe we can chalk Issue 2 up to my misunderstanding?

Issue 1 was fixed in the previous master, it just didn't make it into stable. I would imagine it's just as fixed in the new master.

mathfac, 2020-01-10:
@smorad thank you for flagging it. I updated `stable` tag to include the fix. It should fixed now.

mathfac, 2020-02-06:
Closing as the issue looks resolved.

closed 2020-02-06 by mathfac