facebookresearch/habitat-lab #268: Docker image is out of date.

ZezhouCheng, 2019-12-28:
## ❓ Questions and Help

The current version of habitat in the docker image seems still 0.1.0. Could you update it?

Adidushi, 2020-03-19:
I wrote a [more updated one](https://gist.github.com/Adidushi/bf2251dde6d165fb6e371f7533763729) by changing the commit hashes in the checkout lines (45 & 49).
The current versions are 0.1.4, but you can change the hashes to fit any later versions upon release.
Working alright for me so far.

dhruvbatra, 2020-03-19:
Thanks @Adidushi! If you're like, please feel free to send that dockerfile as a PR to habitat-API.

Adidushi, 2020-03-21:
I'd love to but there seems to be an issue in setup.py where it's not finding an installation of numpy and failing.. Feel free to build the Dockerfile I supplied

Adidushi, 2020-03-22:
More detailed log of the error:
```
The variable 'package_version' was not present in the environment
zip_safe flag not set; analyzing archive contents...
quaternion.__pycache__.numpy_quaternion.cpython-36: module references __file__
removing '/opt/conda/envs/habitat/lib/python3.6/site-packages/numpy_quaternion-2019.12.11.22.25.52-py3.6-linux-x86_64.egg' (and everything under it)
creating /opt/conda/envs/habitat/lib/python3.6/site-packages/numpy_quaternion-2019.12.11.22.25.52-py3.6-linux-x86_64.egg
Extracting numpy_quaternion-2019.12.11.22.25.52-py3.6-linux-x86_64.egg to /opt/conda/envs/habitat/lib/python3.6/site-packages
numpy-quaternion 2019.12.11.22.25.52 is already the active version in easy-install.pth

Installed /opt/conda/envs/habitat/lib/python3.6/site-packages/numpy_quaternion-2019.12.11.22.25.52-py3.6-linux-x86_64.egg
error: The 'numpy' distribution was not found and is required by habitat-sim
The command '/bin/sh -c /bin/bash -c ". activate habitat; cd habitat-sim; git checkout 00cddeee6d66108e277dc5b36006bf3eab03a291; python setup.py build_ext --parallel 2 install"' returned a non-zero code: 1
```

rpartsey, 2022-09-01:
Stale.

closed 2022-09-01 by rpartsey