facebookresearch/habitat-lab #269: Installation of Habitat-API

vamsidesu5, 2019-12-29:
## ❓ Questions and Help

I'm currently trying to install habitat-api and run the example.py script. When I try to install habitat-api, I try running the example script using python examples/example.py and I received the following error: 

Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/Users/vamsi/habitat-api/habitat/core/env.py", line 74, in __init__
    id_dataset=config.DATASET.TYPE, config=config.DATASET
  File "/Users/vamsi/habitat-api/habitat/datasets/registration.py", line 16, in make_dataset
    assert _dataset is not None, "Could not find dataset {}".format(id_dataset)
AssertionError: Could not find dataset PointNav-v1

It seems that it cannot find the dataset in the habitat-api folder. I have been following the installation guide on the README and have placed the data in the correct folder(in habitat-api folder), but I still get this error. 

I have been using these instructions: https://github.com/facebookresearch/habitat-api#installation

Please let me know if I am missing something trivial or if there is another way go about this.

wannabeOG, 2020-01-08:
Check issue #267. There is a conflict between the stable and the master version of the repository.

mathfac, 2020-01-09:
@vamsidesu5, do you still have this issue? Can you `pip uninstall .` in habitat-api folder and check path of `habitat.__path__`? Thanks!

mathfac, 2020-02-06:
Closing as the issue should be resolved.

closed 2020-02-06 by mathfac