facebookresearch/habitat-lab #274: How to apply velocity to the agent?

pushkalkatara, 2020-01-10:
Hi,

I have a vector of linear and angular velocities which I need to apply on the agent at each step. Currently, I am reading the state from `agent.get_state()` and updating the `state.position`, `state.rotation` and `state.sensor_states` in the loop using `agent.set_state()`. Is this the right way or i have to change some other parameters too?

mathfac, 2020-01-10:
Hi @pushkalkatara,
That's one way to go, but that has major drawback that simulator doesn't change for collisions properly for you in this case, as you're teleporting agent.
Correct way to go is use a kinematic functionality of habitat-sim showcased in [this test.](https://github.com/facebookresearch/habitat-sim/blob/master/tests/test_physics.py#L82)

pushkalkatara, 2020-01-10:
Thanks for pointing me to the link.

pushkalkatara, 2020-03-20:
@mathfac `set_translation` and `set_rotation` is not working for the agent. Are these functions only for the objects and not for the agent? My agent id is 0 and no objects are present in the environment.

I tried this controller:
```
def controller(self, agent, velocity):
        current_state=agent.get_state()
        print("OLD state")
        print(current_state)
        self._sim.set_translation(np.array([0.0, 0.0, 0.0]), 0)
        Q = quat_from_angle_axis(np.pi, np.array([0.523599, 0.523599, 0.523599]))
        self._sim.set_rotation(quat_to_magnum(Q), 0)
        current_state=agent.get_state()
        print("New state")
        print(current_state)
```
Agent state
```
OLD state
AgentState(position=array([-1.,  0.,  8.], dtype=float32), rotation=quaternion(1, 0, 0, 0), velocity=array([0., 0., 0.]), angular_velocity=array([0., 0., 0.]), force=array([0., 0., 0.]), torque=array([0., 0., 0.]), sensor_states={'color_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'depth_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0))})
New state
AgentState(position=array([-1.,  0.,  8.], dtype=float32), rotation=quaternion(1, 0, 0, 0), velocity=array([0., 0., 0.]), angular_velocity=array([0., 0., 0.]), force=array([0., 0., 0.]), torque=array([0., 0., 0.]), sensor_states={'color_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'depth_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0))})
```
 

I also tried with the following controller:
```
def agent_controller(self, agent, velocity):
        #_X_AXIS = 0
        #_Y_AXIS = 1
        #_Z_AXIS = 2
        print(velocity)
        axises = [0,1,2]
        current_state=agent.get_state()
        print("OLD state")
        print(current_state)
        for axis in axises:
            # Translate
            ax = agent.scene_node.transformation[axis].xyz
            agent.scene_node.translate_local(ax * velocity[axis])
            # Rotate
            _rotate_local_fns = [
                SceneNode.rotate_x_local,
                SceneNode.rotate_y_local,
                SceneNode.rotate_z_local,
            ]
            #SceneNode.rotate_x_local(agent.scene_node, mn.Deg(velocity[axis)
            _rotate_local_fns[axis](agent.scene_node, mn.Deg(velocity[axis+2]))
            agent.scene_node.rotation = agent.scene_node.rotation.normalized()
        
        current_state=agent.get_state()
        print("OLD state")
        print(current_state)
```

the output was:
```
OLD state
AgentState(position=array([-1.,  0.,  8.], dtype=float32), rotation=quaternion(1, 0, 0, 0), velocity=array([0., 0., 0.]), angular_velocity=array([0., 0., 0.]), force=array([0., 0., 0.]), torque=array([0., 0., 0.]), sensor_states={'color_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'depth_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0))})
NEW state
AgentState(position=array([-1.,  0.,  8.], dtype=float32), rotation=quaternion(0.933012723922729, 0.0669873058795929, 0.250000029802322, 0.25), velocity=array([0., 0., 0.]), angular_velocity=array([0., 0., 0.]), force=array([0., 0., 0.]), torque=array([0., 0., 0.]), sensor_states={'color_sensor': SixDOFPose(position=array([-1.649519,  1.299038,  8.375   ], dtype=float32), rotation=quaternion(0.933012723922729, 0.0669873058795929, 0.250000029802322, 0.25)), 'depth_sensor': SixDOFPose(position=array([-1.649519,  1.299038,  8.375   ], dtype=float32), rotation=quaternion(0.933012723922729, 0.0669873058795929, 0.250000029802322, 0.25))})
```
with this controller agent is rotating but the magnitude is incorrect. I found out this by looking at the quaternion. The quaternion after the controller was 
```
rotation=quaternion(0.933012723922729, 0.0669873058795929, 0.250000029802322, 0.25)
```

but if I consider the ground truth quaternion after rotation of agent by 30 degrees (0.523599 radians),
it should be
```
[ 0.2529195, 0.2529195, 0.2529195, 0.8989412 ]
```

closed 2020-04-01 by pushkalkatara