facebookresearch/habitat-lab #277: How to enable physics when initialising the environment class ?

ronjamoller, 2020-01-17:
I am working on a setup that requires inserted objects so i need the physics_enabled flag and i can get it to run if i set up the simulation myself. However, i have not found a way to set up an environment in such a way that the simulation that it owns has physics enabled. From what i can gather there seems to be no physics flag in the environment config and i have been unsuccessful in reinitializing the env._sim._sim to include physics as well. Is there something that i'm missing ?
Thank you

closed 2020-01-22 by ronjamoller

smorad, 2020-01-29:
This is relevant to me. Is there a way to enable the physics sim via config?

ronjamoller, 2020-01-29:
I have not found a way to do this correctly, my current solution is a workaround: 
env = habitat.Env(config=config)
env._sim._sim.reconfigure(sim_cfg)
that is pretty much the thing i talked about in the above post - you initialize the environment normally (config, see habitat api demo from the habitat docs page)  and then reinitialize the internal simulation using a simulation configuration (sim_cfg, there are examples i think in the habitat-sim demo on the habitat doc page and in the example/demo_runner in the example folder of habitat-sim). My main issue was that there is no consistency in terms of naming - for example in one config the camera might be called rgb, in another it is called rgb_sensor so if you get naming mismatches during the reinitialization check if you can for example rename keys in you sim_config dict to match up with what the environment config expects. I highly suspect that i am using a very hacky solution and while i can run the env and walk around in it and insert objects i have not trained with it yet, so i cannot guarantee anything. Also make sure to pull the newest release, there was quite a change. best of luck :)

smorad, 2020-01-29:
Great, I'd like to patch it correctly. Will hopefully submit a patch to add sim options to the API config tonight.