facebookresearch/habitat-lab #281: RGB Agent

saketd403, 2020-01-23:
## ❓ Questions and Help
Hi,
    I am trying to replicate the results from Habitat paper for various modes of inputs such as RGBD, RGB, Blind. The agent is initially configured for RGBD input. It works well for RGBD input. In order to change the input to RGB, I changed the ppo_pointnav.yaml config file attribute as SENSORS:["RGB_SENSOR"], hence removing the depth sensor. But this causes problem in training the ppo agent as the reward is stuck at -0.009766 and is not improving. Does anything else needs to be changed to configure RGB input?  Also, please let me know the hyperparameters for obtaining the results in Habitat paper for RGB input.

Thanks,
Saurabh

saketd403, 2020-01-25:
Hi,
When I run this with new config file, the code still does not perform well for RGB input. Can you tell me is there something that I am doing wrong while trying to configure the code for getting RGB input? I have configured the code for getting depth as the only input and it worked well.

Thanks,
Saurabh

erikwijmans, 2020-01-25:
It looks like the modified for 1 GPU hyperparameters don't work for RGB unfortunately.  Changing the number of processes to 8 and the number of mini batches to 4 (as in the paper) makes things work for me.

saketd403, 2020-01-26:
Okay. I will try that. By the way in order to get RGB input you just need to change "SENSORS" field in config right? That's how I am doing it.
Thanks,
Saurabh

erikwijmans, 2020-01-26:
Yes, that’s correct.

closed 2020-02-01 by saketd403