facebookresearch/habitat-lab #287: Cannot run Example: No module named 'habitat_sim._ext'

sidwa, 2020-01-29:
I first installed Habitat-sim for cuda and then installed Habitat-api. Habitat-sim works based on the test given in the readme.md. However, running example/example.py in Habitat-api produces the following output:
```
(habitat) siddhant@mobile-desktop:~/libs/habitat-api$ python3 examples/example.py 
2020-01-29 17:27:20,544 Initializing dataset PointNav-v1
Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/home/siddhant/libs/habitat-api/habitat/core/env.py", line 74, in __init__
    id_dataset=config.DATASET.TYPE, config=config.DATASET
  File "/home/siddhant/libs/habitat-api/habitat/datasets/registration.py", line 19, in make_dataset
    return _dataset(**kwargs)
  File "/home/siddhant/libs/habitat-api/habitat/datasets/pointnav/__init__.py", line 38, in __init__
    raise pointnav_import_error
  File "/home/siddhant/libs/habitat-api/habitat/datasets/pointnav/__init__.py", line 20, in _try_register_pointnavdatasetv1
    from habitat.datasets.pointnav.pointnav_dataset import (
  File "/home/siddhant/libs/habitat-api/habitat/datasets/pointnav/pointnav_dataset.py", line 15, in <module>
    from habitat.tasks.nav.nav import (
  File "/home/siddhant/libs/habitat-api/habitat/tasks/nav/nav.py", line 29, in <module>
    from habitat.sims.habitat_simulator.actions import HabitatSimActions
  File "/home/siddhant/libs/habitat-api/habitat/sims/habitat_simulator/actions.py", line 12, in <module>
    import habitat_sim
  File "/home/siddhant/libs/habitat-sim/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/home/siddhant/libs/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ModuleNotFoundError: No module named 'habitat_sim._ext'
```

wannabeOG, 2020-01-30:
Hi, I had a similar issue when I tried installing habitat-API and habitat-sim. 

Make sure that habitat-sim isn't in your PYTHONPATH. The initial parts of this [closed issue](https://github.com/facebookresearch/habitat-sim/issues/316) could be useful

mathfac, 2020-01-30:
Thank you @wannabeOG for helping here! @sidwa, let us know, if the issue is resolved. Smoke test if Habitat SIM was installed correctly is run `python -c "import habitat_sim"` command from any dir except `habitat-sim` folder.

sidwa, 2020-01-30:
Thank you! @wannabeOG I removed the habitat-sim path and then it turned out that I hadn't installed pytorch before since I was on a new virtual env.

closed 2020-01-30 by sidwa