facebookresearch/habitat-lab #288: 'allow_sliding' error when running example.py

qianLyu, 2020-01-30:
Hi,
I'm trying to run: `/habitat-api$ python examples/example.py` after I installed habitat-sim and habitat-api following the instructions. However, there is a bug as follows:

```
2020-01-30 02:12:27,551 Initializing dataset PointNav-v1
2020-01-30 02:12:27,712 initializing sim Sim-v0
Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/home/qian/Git_download/habitat-api/habitat/core/env.py", line 96, in __init__
    id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
  File "/home/qian/Git_download/habitat-api/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/home/qian/Git_download/habitat-api/habitat/sims/habitat_simulator/habitat_simulator.py", line 166, in __init__
    self.sim_config = self.create_sim_config(self._sensor_suite)
  File "/home/qian/Git_download/habitat-api/habitat/sims/habitat_simulator/habitat_simulator.py", line 180, in create_sim_config
    sim_config.allow_sliding = self.config.HABITAT_SIM_V0.ALLOW_SLIDING
AttributeError: 'habitat_sim._ext.habitat_sim_bindings.SimulatorConfiguration' object has no attribute 'allow_sliding'
```

It seems that there's some problem within 'allow_sliding'. Could you help me solve the problem? Thanks a lot!

mathfac, 2020-01-30:
Try to pull latest Habitat-API changes, specifically [this commit](https://github.com/facebookresearch/habitat-api/commit/1ff893a75f0623a4b0cf93a8d3cee1e122c0a623) is required for compatibility.

qianLyu, 2020-01-30:
> Try to pull latest Habitat-API changes, specifically [this commit](https://github.com/facebookresearch/habitat-api/commit/1ff893a75f0623a4b0cf93a8d3cee1e122c0a623) is required for compatibility.

Thanks for your help! I'm finally able to run the code after I update habitat-sim.

closed 2020-02-06 by mathfac

tfy14esa, 2020-02-14:
I am getting the exact same problem as stated above, but it seems like I have already the most recent version of the habitat-api. The first two commits when running 'git log' are:

`commit e368e3a17f62d08bee92afb9d316515d1fd7a7a4
Author: Oleksandr <maksymets.o@gmail.com>
Date:   Wed Jan 29 09:46:48 2020 -0800

    Version bump, better late than never (#286)
    
    Version bump, better late than never. That's needed for right package version building.

commit 1ff893a75f0623a4b0cf93a8d3cee1e122c0a623
Author: Erik Wijmans <etw@gatech.edu>
Date:   Mon Jan 27 20:32:40 2020 -0500

    Interface with habitat-sim no sliding (#283)
    
    Habitat sim can now disable sliding. This PR wires up habitat-api with that.
`

I installed habitat-api successfully according to the build instructions. I installed habitat-sim on my Linux machine (output from cat /proc/version):
`Linux version 4.9.0-11-amd64 (debian-kernel@lists.debian.org) (gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1) ) #1 SMP Debian 4.9.189-3+deb9u2 (2019-11-11)`

I ran the commands following [this](https://github.com/facebookresearch/habitat-sim/pull/445):
```
conda install -c aihabitat -c conda-forge headless
conda install -c aihabitat -c conda-forge habitat-sim
```

erikwijmans, 2020-02-14:
The conda binaries are currently in pre-alpha and aren't synced for the release.  I am not sure what commit those were built for, but it is likely that they don't have this feature.