facebookresearch/habitat-lab #292: Trouble calling fucntions of RLEnv from VectorEnv class

saimwani, 2020-02-04:
## ❓ Questions and Help
Hi,
I am trying to get the GPS coordinates of an agent in an env of VectorEnv class. Since I believe I would need the ._sim variable for the purpose, I call self._connection_write_fns[i](("habitat_env", None)) on VectorEnv, which should call the 'habitat_env' function on the RLEnv with index i. This doesn't seem to work as I get an EOF error on running self._connection_read_fns[i]() to read the result. I can, however, call some other functions of the RLEnv in the same way, such as "current_episode" and "reset".
Could you please help me resolve this? Thanks a lot!

erikwijmans, 2020-02-06:
`habitat_env` is not pickle-able, so this will never work unfortunately.

Please look at the existing sensors to see if any of those fit your need, if not, you can add an additional sensor easily.  Sensors and measurements are the best way to expose things like this.

https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav.py

https://github.com/facebookresearch/habitat-api/blob/master/examples/register_new_sensors_and_measures.py

closed 2020-02-06 by erikwijmans

AAdeleye, 2020-08-04:
Hello, I have a question similar to this. 

I want to use the 'env.sim.semantic_annotations()' method listed [here](https://aihabitat.org/docs/habitat-api/habitat-api-demo.html).

I am trying to do something along the lines of `self.envs._connection_write_fns[0](??, none)` to get a single env.  I don't think this is possible anymore so I was wondering how best to go about this? 

I wanted the semantic_annotations after a signal step (i.e to change the following lines `self.envs_connection_write_fns[0](STEP_COMMAND, action)` `self.env_connection_read_fns[0]()` so that read includes semantic_annotations)

Thank you.

saimwani, 2020-08-07:
You can add a new measure for this as shown [here](https://github.com/facebookresearch/habitat-api/blob/master/examples/register_new_sensors_and_measures.py). This will return the desired measurement after every step for each of the environments.