facebookresearch/habitat-lab #293: Generating my own dataset

smorad, 2020-02-05:
## ❓ Questions and Help
I only want to train on `easy` difficulty datasets. From what I can see, there isn't a way to do this easily using the API. I'm willing to generate a new dataset using `pointnav_generator.py`, but I don't understand how to convert/export it from a python `Object` to a `json.gz` like the other premade gibson datasets.

mathfac, 2020-02-06:
To train on `easy` difficulty datasets you can use [filter_episode function](https://github.com/facebookresearch/habitat-api/blob/master/habitat/core/dataset.py#L136) to leave only easy episodes.

Regarding saving dataset to the disk: after creating a dataset with episodes you can save it on the disk: 
```
import gzip
...
with gzip.GzipFile(main_dataset_file, "wb") as f:
  f.write(dataset.to_json().encode("utf-8"))
```

tfy14esa, 2020-02-11:
Hi @mathfac ,
This is a follow-up question related to the topic "Generating my own dataset". Does Habitat give you the functionality to render your own dataset with a user-specified trajectory and sensor suite? 

More concretely, I am looking for something similar to the ViSim by [InteriorNet](https://interiornet.org/) (the ViSim is not released yet). Perhaps an equivalent setup does not exist in Habitat, but maybe the API offers the same tools?

Any help is appreciated.

smorad, 2020-02-11:
So I can generate datasets, but is there a way for me to specify a specific `.glb` mesh to load?

To actually use `generate_pointnav_episode`, I need a specific mesh to be loaded into the `sim`. There doesn't seem to be a way to control this. I'd like to tell a sim to load `xxx.glb`. 

If you launch a sim and iterate through the episodes forever, certain meshes never appear. `Euharlee.glb`, for example. Even if I set `DATASET.SCENES_DIR`, in the `yaml` it seems to try and load meshes that do not exist in that directory.

```
cfg = habitat.get_config('episode_generator.yaml') 
# yaml contains: SCENE_DIR=...gibson
# directory structure
... gibson|-> Euharlee.glb
          |-> Euharlee.navmesh
env = habitat.Env(cfg)
```

```
ValueError: Cannot load: .../gibson/Dunmor.glb 
```

tfy14esa, 2020-03-04:
Hi @smorad,
I am also trying to create my own dataset. By this I mean that I want to store the rgb frames to disk as I traverse the scene. I am able to do this by navigating with the keyboard, but this is tedious. A faster way would be to have an agent walk the shortest geodesic path from some random starting position to some random goal position. 

Were you able to do this, and if so, could you please provide me with the necessary code?

Cheers!

smorad, 2020-03-04:
Unfortunately I wasn't able to figure out how to load a specific scene. Would be interested to know if you get it working.

tfy14esa, 2020-03-04:
The way I load a specific scene is similar to what is done in [this](https://aihabitat.org/docs/habitat-sim/stereo-agent.html) example code. Note that this example code does not define a task and episodes but only uses habitat_sim.

Is this helpful to you?

ronjamoller, 2020-03-05:
Hi, i too would like to create my own dataset but with my own .glb files. I have looked into test_pointnav_dataset, pointnav_generator and the run.py for the baseline examples. However, i am still unsure how to achieve my goal - my understanding is that i need to first create the jsons (train, val, test) that determine what .glbs are loaded and set the start and goal positions. The only way the get the jsons for the existing supported datasets seems to be downloading them, is there a tool to create the json ? There is a similar issue about exactly this problem with the replica dataset  ([https://github.com/facebookresearch/habitat-api/issues/136])but as far as i can tell there is no answer yet.

rosanom, 2020-03-05:
Hi, I join the thread because I have the same need. I have a custom environment and I have to generate the json episodes file to later train and evaluate a RL model. What is the correct procedure to create it with the simulator api? Can somebody provide even a short code example?
Thanks

rosanom, 2020-03-09:
I've found a jupyter notebook by erikwijmans that shows exactly how to generate the navigation episodes given the environment mesh (glb): [erikwijmans/create_pointnav_dataset.py](https://gist.github.com/erikwijmans/1c596183fb8de1e9afd80088b6b5c115)
Hope it helps!

closed 2023-01-26 by rpartsey