facebookresearch/habitat-lab #296: Right way to specify GPU memory in DDPPO.

pushkalkatara, 2020-02-16:
Hi,

What parameters do we need to change in order to utilize all GPU resources?

I'm reproducing the DDPPO results using the `single_node.sh` script. I have a GPU with 12GB of VRAM as below.
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 390.116                Driver Version: 390.116                   |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX TIT...  Off  | 00000000:05:00.0 Off |                  N/A |
| 22%   59C    P8    30W / 250W |   4501MiB / 12210MiB |     11%      Default |
+-------------------------------+----------------------+----------------------+
```

The VRAM utilized is only 4GB. There are 4 workers running,
```
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0     17009      G   ...a/miniconda3/envs/habitatapi/bin/python  1382MiB |
|    0     17010      G   ...a/miniconda3/envs/habitatapi/bin/python  1375MiB |
|    0     17011      G   ...a/miniconda3/envs/habitatapi/bin/python  1377MiB |
|    0     17012      G   ...a/miniconda3/envs/habitatapi/bin/python   355MiB |
+-----------------------------------------------------------------------------+
```

How can I utilize the remaining 8GBs to speed up the training? In the [ddppo_pointnav.yaml](https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/config/pointnav/ddppo_pointnav.yaml), NUM_PROCESSES = 4. Should I increase this param or there are other configurations I need to change as well?

erikwijmans, 2020-02-16:
Yes, you can increase the number of processes.  With that said, I am surprised that there is no GPU memory being used by Cuda (would be type C), is the model on the CPU?  In which case, I would highly recommend using the remaining GPU memory for the model and forward/backward passes.

pushkalkatara, 2020-02-17:
Thanks for the response. I would increase the number of workers if required. 

Yes, I also noticed the model being loaded on the CPU. I installed pytorch with cudatoolkit. Weird the model loaded on CPU. Any ideas why?

erikwijmans, 2020-02-18:
Habitat-baselines will fall back to using the CPU if CUDA is not detected. You can check to see if pytorch is loading CUDA correctly via `python -c "import torch; print(torch.cuda.is_available())"`.  If that returns False, either pytorch or CUDA is not setup correctly.  Your nvidia-smi output is lacking a CUDA Version, suggesting that CUDA is not installed correctly on your machine.

pushkalkatara, 2020-02-18:
Old nvidia drivers didn't show CUDA version in nvidia-smi. [>=410.72](https://devtalk.nvidia.com/default/topic/1045528/cuda-setup-and-installation/nvidia-smi-doesn-t-show-cuda-version-even-after-installation/). nvcc --version is returning as CUDA installed:
```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2018 NVIDIA Corporation
Built on Sat_Aug_25_21:08:01_CDT_2018
Cuda compilation tools, release 10.0, V10.0.130
```

thanks for pointing out the command, `torch.cuda.is_available()` is returning False. Tensorflow is reading the the GPU as `tf.test.is_gpu_available()` is returning True. I think there is a CUDA version conflict for installed Pytorch.

rosanom, 2020-03-19:
Hello,
following the discussion about the correct load of the model on the GPU and the use of memory by CUDA, I installed `habitat-sim` on a headless machine with a `1 x K80 GPU` and `CUDA 10.2` using the `--with-cuda` and `--headless` flags and I run the ppo and ddppo baselines with the default configuration on the skokloster scene (`SENSORS: ["RGB_SENSOR", "DEPTH_SENSOR"]`) . Running the `nvidia-smi` command I get the following active processes:
for PPO model:
<pre>+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0     13548      C   python                                      1475MiB |
|    0     13656      G   .../anaconda3/envs/envname/bin/python        374MiB |
+-----------------------------------------------------------------------------+
</pre>
for DDPPO model:
<pre>+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0     32504      C   .../anaconda3/envs/envname/bin/python       7108MiB |
|    0     32631      G   .../anaconda3/envs/envname/bin/python        246MiB |
|    0     32632      G   .../anaconda3/envs/envname/bin/python        246MiB |
|    0     32633      G   .../anaconda3/envs/envname/bin/python        373MiB |
|    0     32634      G   .../anaconda3/envs/envname/bin/python        246MiB |
+-----------------------------------------------------------------------------+
</pre>
For both models I'm getting a framerate of around `50fps` and it seems to me a bit low. 
Moreover, running the `habitat-sim/examples/benchmark.py` on the same scene I can report this output:
<pre>[EGL] Detected 1 EGL devices
[EGL] Selected EGL device 0 for CUDA device 0
[EGL] Version: 1.5
[EGL] Vendor: NVIDIA
Renderer: Tesla K80/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.33.01
...
====== FPS (128 x 128, rgb): 1005.5 ======
====== FPS (128 x 128, rgbd): 449.5 ======
====== FPS (128 x 128, depth_only): 829.3 ======
====== FPS (256 x 256, rgb): 941.5 ======
====== FPS (256 x 256, rgbd): 371.3 ======
====== FPS (256 x 256, depth_only): 773.2 ======
...
</pre>
Is it what I should expect or perhaps something is going wrong?

Thank you.

erikwijmans, 2020-03-19:
This all seems correct to me.

> For both models I'm getting a framerate of around 50fps and it seems to me a bit low.

The frame rates reported by the training script include the simulation time, the model inference time, and the parameter update time, so it will be considerably lower than just the simulator itself.

rpartsey, 2022-09-01:
All the questions were answered in the Issue comments.

closed 2022-09-01 by rpartsey