facebookresearch/habitat-lab #298: Segfault and core dump when run multiple-node DDPPO using Slurm

Shigangli, 2020-02-18:
## ❓ Questions and Help
Hi,

I have successfully run PPO using the MatterPort3D dataset. Currently I'm trying to run multiple-node DDPPO using `multi_node_slurm.sh`. But I got a very strange problem. Whenever I use `srun` and import some modules (e.g. `from habitat_baselines.common.baseline_registry import baseline_registry` `from habitat_baselines.config.default import get_config`) from Habitat, it aborted with Segfault as follows.

`srun: error: nid04277: task 1: Segmentation fault (core dumped)
srun: Terminating job step 20370799.1
srun: error: nid04276: task 0: Segmentation fault (core dumped)
`

However, if I don't use `srun` (namely using `python xxx.py` to run one a single node), there is no Segfault. I got stuck by this for several days and frustrated. Is there any clue to solve this problem?

erikwijmans, 2020-02-18:
I have never seen this issue :/

Skylion007, 2020-02-19:
Modify the slurm script by 

> -X faulthandler

 as the first python arg. This will give a stack trace of where the segfault is occuring.

Shigangli, 2020-03-04:
Thanks for your comments.

I managed to solve this problem by rebuilding it.

closed 2020-03-04 by Shigangli