facebookresearch/habitat-lab #300: Using pre-trained models with habitat_baselines

PrieureDeSion, 2020-02-19:
How does one use the pre-trained models (linked [here](https://github.com/facebookresearch/habitat-api/tree/master/habitat_baselines)) with the habitat_baselines `run.py` script?

Just treating them as eval checkpoints does not seem to be the right way due to the difference in structure.

PrieureDeSion, 2020-02-19:
As a temporary fix, I modified the keys in the state_dict to reflect the changes in name (assuming the structure of the Policy network has remained unchanged otherwise).
Yet, the pre-trained model looks pretty bad and has definitely not been loaded right (attached some rollout videos).

[image]
[image]

erikwijmans, 2020-02-19:
The actions have change what number they are on, here is a little script I made to fix the checkpoints; https://gist.github.com/erikwijmans/94f33461982ac9e3cc354fb26e4b80e1

closed 2020-02-19 by erikwijmans

PrieureDeSion, 2020-02-21:
@erikwijmans is there information on how the pre-trained baselines were trained, i.e. on Gibson or MP3D training split etc.?

erikwijmans, 2020-02-21:
On Gibson.