facebookresearch/habitat-lab #301: PointNav training curriculum

PrieureDeSion, 2020-02-19:
The default PointNav task for MP3D seems very hard given how far the sampled goals can get. Is there a principled/already implemented way to sample goals in a smaller neighborhood and specify as a config parameter?
If there isn't, any references on where I could start off to implement something like that or a curriculum of increasing distances?

ronjamoller, 2020-03-24:
I am interested in curriculum learning as well

rpartsey, 2022-09-01:
Suggestions on how to filter/implement a custom episodes order:
- generate your own PointNav task dataset with a short geodesic distance to target
- use [Dataset.filter_episodes](https://github.com/facebookresearch/habitat-lab/blob/cd84d56b351b7389a24c16b785b9bc141e078858/habitat/core/dataset.py#L223) method to filter episodes
- define a custom [EpisodeIterator](https://github.com/facebookresearch/habitat-lab/blob/cd84d56b351b7389a24c16b785b9bc141e078858/habitat/core/dataset.py#L336) and use it in [Dataset.get_episode_iterator](https://github.com/facebookresearch/habitat-lab/blob/cd84d56b351b7389a24c16b785b9bc141e078858/habitat/core/dataset.py#L180)

closed 2022-09-01 by rpartsey