facebookresearch/habitat-lab #302: CUDA RuntimeError on training baseline PPO

PrieureDeSion, 2020-02-19:
## 🐛 Bug
On training the PPO baseline, the trainer crashes every once in a while due to a `RuntimeError: CUDA error: device-side assert triggered`. Looks like it occurs when the reward is `NaN`, but I am not certain. The error message is shown below for reference:

```
I0219 00:00:30.755550 28865 ResourceManager.cpp:1054] Importing Basis files as BC7
I0219 00:00:40.866313 28865 Simulator.cpp:112] Loading house from data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.house
I0219 00:00:40.866335 28865 Simulator.cpp:118] Loading semantic mesh data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH_semantic.ply
I0219 00:00:41.459719 28865 Simulator.cpp:130] Loaded.
I0219 00:00:41.475843 28865 simulator.py:142] Loaded navmesh data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.navmesh
2020-02-19 00:00:46,051 update: 220     fps: 91.648
2020-02-19 00:00:46,051 update: 220     env-time: 193.507s      pth-time: 113.527s      frames: 28288
2020-02-19 00:00:46,051 Average window size 50 reward: nan
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [0,0
,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [1,0
,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [2,0
,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [3,0
,0] Assertion `val >= zero` failed.
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 68, in <module>
    main()
  File "habitat_baselines/run.py", line 38, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 62, in run_exp
    trainer.train()
  File "<...>/habitat-api/habitat_baselines/rl/ppo/ppo_trainer.py", line 300, in train
    episode_counts,
  File "<...>/habitat-api/habitat_baselines/rl/ppo/ppo_trainer.py", line 146, in _collect_rollout_step
    outputs = self.envs.step([a[0].item() for a in actions])
  File "<...>/habitat-api/habitat_baselines/rl/ppo/ppo_trainer.py", line 146, in <listcomp>
    outputs = self.envs.step([a[0].item() for a in actions])
RuntimeError: CUDA error: device-side assert triggered
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7ff7b85f16a0>>
Traceback (most recent call last):
  File "<...>/habitat-api/habitat/core/vector_env.py", line 468, in __del__
  File "<...>/habitat-api/habitat/core/vector_env.py", line 350, in close
  File "<...>/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 206, in send
AttributeError: 'NoneType' object has no attribute 'dumps'
```

## Command
`python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav.yaml --run-type train`

pushkalkatara, 2020-03-09:
Facing the same issue. any resolution yet?

ybgdgh, 2020-09-11:
The same issue.

erikwijmans, 2020-09-11:
This should be fixed.  There were a couple episodes of PointNav in the mp3d train dataset that became unnavigable after a seemingly unrelated change.  Those have been removed.  Please re-download the pointnav episodes.

rpartsey, 2022-09-01:
Closing the issue because it has not had recent activity.

closed 2022-09-01 by rpartsey