facebookresearch/habitat-lab #304: How to run objectnav baseline?

qianLyu, 2020-02-19:
Hi all,

I've updated latest version of both habitat-api and habitat-sim and try to train object-nav baseline by simply running the code as follows:
`python -u habitat_baselines/run.py --exp-config habitat_baselines/config/objectnav/ddppo_objectnav.yaml --run-type train`

However, after showing config parameters, the programme seems to stuck in this place:
`Initializing dataset ObjectNav-v1`

I've properly installed mp3d and its dataset, but I run on a single gpu and I don't have NCCL. Is that the cause? And what is the right procedure to run objectnav baseline?

Thank you!

erikwijmans, 2020-02-19:
You can change NCCL to GLOO here: https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/config/objectnav/ddppo_objectnav.yaml#L45

@mathfac we should make a policy registry, that way people can just the PPO if they only want to use 1 GPU

qianLyu, 2020-02-20:
> You can change NCCL to GLOO here: https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/config/objectnav/ddppo_objectnav.yaml#L45
> 
> @mathfac we should make a policy registry, that way people can just the PPO if they only want to use 1 GPU

qianLyu, 2020-02-20:
@erikwijmans Thank you! I tried to run ddppo on pointnav baseline on a single gpu, and it turned out to be good. However, when I just changed the dataset to mp3d(no other modification), the programme stuck and reported error: 
`BrokenPipeError: [Errno 32] Broken pipe `
I guess it seems like a hardware problem. To run ddppo on mp3d dataset, maybe it's better to use multiple gpus or a better type of gpu. I'll try this out. Btw, I run the mp3d baseline on my notebook with GTX1060 Maxq.

qianLyu, 2020-02-20:
And I downloaded NCCL. It supports the use of single gpu.

mathfac, 2020-05-19:
@qianLyu `BrokenPipeError: [Errno 32] Broken pipe` is usually cause by other error with environment running as separate processes. Try to check memory occupation on the GPU as well as decrease number of processes. MP3D 3D scenes take more memory than Gibson's ones.

rpartsey, 2022-09-01:
Closing the issue because it has not had recent activity.

closed 2022-09-01 by rpartsey