facebookresearch/habitat-lab #307: Out of date Documentation

tfy14esa, 2020-02-20:
## ❓ Questions and Help
Hi,
The documentation to habitat is available [here](https://aihabitat.org/docs/habitat-api/index.html). I wanted to try the quickstart code found [here](https://aihabitat.org/docs/habitat-api/quickstart.html),  however, the following error is given:

```
(/scratch_net/nudel/esandstroem/venvs/habitat_wo_headless_wo_cuda_env) esandstroem@nudel:~/scratch/programs/habitat-api/tutorials$ python pointnav_task.py 
2020-02-20 17:31:22,403 Initializing dataset PointNav-v1
2020-02-20 17:31:22,520 initializing sim Sim-v0
Renderer: GeForce GT 1030/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 390.116
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0220 17:31:22.560645  4263 ResourceManager.cpp:1054] Importing Basis files as BC7
W0220 17:31:23.024389  4263 Simulator.cpp:140] :
---
 The active scene does not contain semantic annotations. 
---
I0220 17:31:23.024598 4263 simulator.py:142] Loaded navmesh /home/esandstroem/scratch/programs/habitat-api/data/scene_datasets/habitat-test-scenes/van-gogh-room.navmesh
2020-02-20 17:31:23,025 Initializing task Nav-v0
Environment creation successful
Destination, distance: 1.166085, theta(radians): -2.09
Agent stepping around inside environment.
Traceback (most recent call last):
  File "pointnav_task.py", line 64, in <module>
    example()
  File "pointnav_task.py", line 33, in example
    action = habitat.SimulatorActions.MOVE_FORWARD
AttributeError: module 'habitat' has no attribute 'SimulatorActions'
I0220 17:31:24.855016  4263 Simulator.cpp:35] Deconstructing Simulator
I0220 17:31:24.855033  4263 SemanticScene.h:40] Deconstructing SemanticScene
I0220 17:31:24.855036  4263 SceneManager.h:24] Deconstructing SceneManager
I0220 17:31:24.855038  4263 SceneGraph.h:20] Deconstructing SceneGraph
I0220 17:31:24.855046  4263 RenderTarget.h:51] Deconstructing RenderTarget
I0220 17:31:24.855113  4263 Sensor.h:80] Deconstructing Sensor
I0220 17:31:24.858453  4263 Renderer.cpp:33] Deconstructing Renderer
I0220 17:31:24.858466  4263 WindowlessContext.h:16] Deconstructing WindowlessContext
I0220 17:31:24.858469  4263 WindowlessContext.cpp:245] Deconstructing GL context
```
The line
```
AttributeError: module 'habitat' has no attribute 'SimulatorActions'
```
is specifically interesting.

I found in the script located at habitat-api/habitat/datasets/pointnav/__init__.py which says the following in the comments:
```
# TODO(akadian): This is a result of moving SimulatorActions away from core
# and into simulators specifically. As a result of that the connection points
# for our tasks and datasets for actions is coming from inside habitat-sim
# which makes it impossible for anyone to use habitat-api without having
# habitat-sim installed. In a future PR we will implement a base simulator
# action class which will be the connection point for tasks and datasets.
# Post that PR we would no longer need try register blocks.
```

**What should I replace "SimulatorActions" with in order to run the code?**

Furthermore, I also had to change the key "pointgoal" to "pointgoal_with_gps_compass".

Thanks.

tfy14esa, 2020-02-21:
I finally found the solution:
1. Add the import
```
from habitat.sims.habitat_simulator.actions import HabitatSimActions
```
Then form the actions as:
```
action = HabitatSimActions.MOVE_FORWARD
```

closed 2020-02-21 by tfy14esa

dhruvbatra, 2020-02-23:
@tfy14esa -- would you mind sending a PR to fix that in the docs?