facebookresearch/habitat-lab #308: Error while running ObjectNav task

shivanshpatel35, 2020-02-20:
## ❓ Help
I am trying to run objectnav using the script `habitat_baseline/rl/ddppo/single_node.sh` on 2 GPU machine. I have edited `--exp-config` flag to `habitat_baselines/config/objectnav/ddppo_objectnav.yaml`.
I am getting the following error log
```2020-02-20 13:40:17,978 config: BASE_TASK_CONFIG_PATH: configs/tasks/objectnav_mp3d.yaml
CHECKPOINT_FOLDER: new_checkpoints
CHECKPOINT_INTERVAL: 50
CMD_TRAILING_OPTS: []
ENV_NAME: NavRLEnv
EVAL:
  SPLIT: val
  USE_CKPT_CONFIG: True
EVAL_CKPT_PATH_DIR: new_checkpoints
LOG_FILE: train.log
LOG_INTERVAL: 10
NUM_PROCESSES: 1
NUM_UPDATES: 10000
ORBSLAM2:
  ANGLE_TH: 0.2617993877991494
  BETA: 100
  CAMERA_HEIGHT: 1.25
  DEPTH_DENORM: 10.0
  DIST_REACHED_TH: 0.15
  DIST_TO_STOP: 0.05
  D_OBSTACLE_MAX: 4.0
  D_OBSTACLE_MIN: 0.1
  H_OBSTACLE_MAX: 1.25
  H_OBSTACLE_MIN: 0.375
  MAP_CELL_SIZE: 0.1
  MAP_SIZE: 40
  MIN_PTS_IN_OBSTACLE: 320.0
  NEXT_WAYPOINT_TH: 0.5
  NUM_ACTIONS: 3
  PLANNER_MAX_STEPS: 500
  PREPROCESS_MAP: True
  SLAM_SETTINGS_PATH: habitat_baselines/slambased/data/mp3d3_small1k.yaml
  SLAM_VOCAB_PATH: habitat_baselines/slambased/data/ORBvoc.txt
RL:
  DDPPO:
    backbone: resnet50
    distrib_backend: NCCL
    num_recurrent_layers: 2
    pretrained: False
    pretrained_encoder: False
    pretrained_weights: data/ddppo-models/gibson-2plus-resnet50.pth
    reset_critic: True
    rnn_type: LSTM
    sync_frac: 0.6
    train_encoder: True
  PPO:
    clip_param: 0.2
    entropy_coef: 0.01
    eps: 1e-05
    gamma: 0.99
    hidden_size: 512
    lr: 2.5e-06
    max_grad_norm: 0.2
    num_mini_batch: 2
    num_steps: 128
    ppo_epoch: 2
    reward_window_size: 50
    tau: 0.95
    use_gae: True
    use_linear_clip_decay: False
    use_linear_lr_decay: False
    use_normalized_advantage: False
    value_loss_coef: 0.5
  REWARD_MEASURE: distance_to_goal
  SLACK_REWARD: -0.01
  SUCCESS_MEASURE: spl
  SUCCESS_REWARD: 2.5
[... 206 lines cut ...]
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7efc2f8b1668>>
Traceback (most recent call last):
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 468, in __del__
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 347, in close
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 250, in recv
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 375, in _recv
AttributeError: 'NoneType' object has no attribute 'BytesIO'
Traceback (most recent call last):
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/site-packages/torch-1.4.0-py3.6-linux-x86_64.egg/torch/distributed/launch.py", line 263, in <module>
    main()
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/site-packages/torch-1.4.0-py3.6-linux-x86_64.egg/torch/distributed/launch.py", line 259, in main
    cmd=cmd)
subprocess.CalledProcessError: Command '['/local-scratch/anaconda3/envs/habitat/bin/python', '-u', 'habitat_baselines/run.py', '--exp-config', 'habitat_baselines/config/objectnav/ddppo_objectnav.yaml', '--run-type', 'train']' returned non-zero exit status 1.
```
Thanks in advance

qianLyu, 2020-02-21:
Hello, I guess you could try to tune 'NUM_PROCESSES' parameter higher, since you have gpu0 and gpu1 running.

erikwijmans, 2020-02-21:
Hi, can you turn logging from the simulator on?  i.e. comment out these two lines: https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/rl/ddppo/single_node.sh#L3-L4

shivanshpatel35, 2020-02-21:
This leads to the following log
```2020-02-21 08:44:06,775 config: BASE_TASK_CONFIG_PATH: configs/tasks/objectnav_mp3d.yaml
CHECKPOINT_FOLDER: new_checkpoints
CHECKPOINT_INTERVAL: 50
CMD_TRAILING_OPTS: []
ENV_NAME: NavRLEnv
EVAL:
  SPLIT: val
  USE_CKPT_CONFIG: True
EVAL_CKPT_PATH_DIR: new_checkpoints
LOG_FILE: train.log
LOG_INTERVAL: 10
NUM_PROCESSES: 1
NUM_UPDATES: 10000
ORBSLAM2:
  ANGLE_TH: 0.2617993877991494
  BETA: 100
  CAMERA_HEIGHT: 1.25
  DEPTH_DENORM: 10.0
  DIST_REACHED_TH: 0.15
  DIST_TO_STOP: 0.05
  D_OBSTACLE_MAX: 4.0
  D_OBSTACLE_MIN: 0.1
  H_OBSTACLE_MAX: 1.25
  H_OBSTACLE_MIN: 0.375
  MAP_CELL_SIZE: 0.1
  MAP_SIZE: 40
  MIN_PTS_IN_OBSTACLE: 320.0
  NEXT_WAYPOINT_TH: 0.5
  NUM_ACTIONS: 3
  PLANNER_MAX_STEPS: 500
  PREPROCESS_MAP: True
  SLAM_SETTINGS_PATH: habitat_baselines/slambased/data/mp3d3_small1k.yaml
  SLAM_VOCAB_PATH: habitat_baselines/slambased/data/ORBvoc.txt
RL:
  DDPPO:
    backbone: resnet50
    distrib_backend: NCCL
    num_recurrent_layers: 2
    pretrained: False
    pretrained_encoder: False
    pretrained_weights: data/ddppo-models/gibson-2plus-resnet50.pth
    reset_critic: True
    rnn_type: LSTM
    sync_frac: 0.6
    train_encoder: True
  PPO:
    clip_param: 0.2
    entropy_coef: 0.01
    eps: 1e-05
    gamma: 0.99
    hidden_size: 512
    lr: 2.5e-06
    max_grad_norm: 0.2
    num_mini_batch: 2
    num_steps: 128
    ppo_epoch: 2
    reward_window_size: 50
    tau: 0.95
    use_gae: True
    use_linear_clip_decay: False
    use_linear_lr_decay: False
    use_normalized_advantage: False
    value_loss_coef: 0.5
  REWARD_MEASURE: distance_to_goal
  SLACK_REWARD: -0.01
  SUCCESS_MEASURE: spl
  SUCCESS_REWARD: 2.5
SENSORS: ['DEPTH_SENSOR', 'RGB_SENSOR']
SIMULATOR_GPU_ID: 0
[... 230 lines cut ...]
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7fd7b6769898>>
Traceback (most recent call last):
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 468, in __del__
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 347, in close
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 250, in recv
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 375, in _recv
AttributeError: 'NoneType' object has no attribute 'BytesIO'
Traceback (most recent call last):
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/site-packages/torch-1.4.0-py3.6-linux-x86_64.egg/torch/distributed/launch.py", line 263, in <module>
    main()
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/site-packages/torch-1.4.0-py3.6-linux-x86_64.egg/torch/distributed/launch.py", line 259, in main
    cmd=cmd)
subprocess.CalledProcessError: Command '['/local-scratch/anaconda3/envs/habitat/bin/python', '-u', 'habitat_baselines/run.py', '--exp-config', 'habitat_baselines/config/objectnav/ddppo_objectnav.yaml', '--run-type', 'train']' returned non-zero exit status 1.```

erikwijmans, 2020-02-21:
I don't see anything from EGL in the logs, indicating that habitat-sim was built without `--headless`?  Habitat-sim needs to be built with `--headless` to leverage multiple GPUs.

shivanshpatel35, 2020-02-21:
I installed Habitat-sim using `--headless` flag. But I reinstalled it in a new environment just for peace of mind. I am still getting the error log -
```2020-02-21 11:54:40,275 config: BASE_TASK_CONFIG_PATH: configs/tasks/objectnav_mp3d.yaml
CHECKPOINT_FOLDER: new_checkpoints
CHECKPOINT_INTERVAL: 50
CMD_TRAILING_OPTS: []
ENV_NAME: NavRLEnv
EVAL:
  SPLIT: val
  USE_CKPT_CONFIG: True
EVAL_CKPT_PATH_DIR: new_checkpoints
LOG_FILE: train.log
LOG_INTERVAL: 10
NUM_PROCESSES: 1
NUM_UPDATES: 10000
ORBSLAM2:
  ANGLE_TH: 0.2617993877991494
  BETA: 100
  CAMERA_HEIGHT: 1.25
  DEPTH_DENORM: 10.0
  DIST_REACHED_TH: 0.15
  DIST_TO_STOP: 0.05
  D_OBSTACLE_MAX: 4.0
  D_OBSTACLE_MIN: 0.1
  H_OBSTACLE_MAX: 1.25
  H_OBSTACLE_MIN: 0.375
  MAP_CELL_SIZE: 0.1
  MAP_SIZE: 40
  MIN_PTS_IN_OBSTACLE: 320.0
  NEXT_WAYPOINT_TH: 0.5
  NUM_ACTIONS: 3
  PLANNER_MAX_STEPS: 500
  PREPROCESS_MAP: True
  SLAM_SETTINGS_PATH: habitat_baselines/slambased/data/mp3d3_small1k.yaml
  SLAM_VOCAB_PATH: habitat_baselines/slambased/data/ORBvoc.txt
RL:
  DDPPO:
    backbone: resnet50
    distrib_backend: NCCL
    num_recurrent_layers: 2
    pretrained: False
    pretrained_encoder: False
    pretrained_weights: data/ddppo-models/gibson-2plus-resnet50.pth
    reset_critic: True
    rnn_type: LSTM
    sync_frac: 0.6
    train_encoder: True
  PPO:
    clip_param: 0.2
    entropy_coef: 0.01
    eps: 1e-05
    gamma: 0.99
    hidden_size: 512
    lr: 2.5e-06
    max_grad_norm: 0.2
    num_mini_batch: 1
    num_steps: 128
    ppo_epoch: 2
    reward_window_size: 50
    tau: 0.95
    use_gae: True
    use_linear_clip_decay: False
    use_linear_lr_decay: False
    use_normalized_advantage: False
    value_loss_coef: 0.5
  REWARD_MEASURE: distance_to_goal
  SLACK_REWARD: -0.01
  SUCCESS_MEASURE: spl
  SUCCESS_REWARD: 2.5
SENSORS: ['DEPTH_SENSOR', 'RGB_SENSOR']
SIMULATOR_GPU_ID: 0
[... 594 lines cut ...]
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7efb2bd01668>>
Traceback (most recent call last):
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 468, in __del__
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 347, in close
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 250, in recv
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/multiprocessing/connection.py", line 375, in _recv
AttributeError: 'NoneType' object has no attribute 'BytesIO'
Traceback (most recent call last):
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/site-packages/torch-1.4.0-py3.6-linux-x86_64.egg/torch/distributed/launch.py", line 263, in <module>
    main()
  File "/local-scratch/anaconda3/envs/habitat2/lib/python3.6/site-packages/torch-1.4.0-py3.6-linux-x86_64.egg/torch/distributed/launch.py", line 259, in main
    cmd=cmd)
subprocess.CalledProcessError: Command '['/local-scratch/anaconda3/envs/habitat2/bin/python', '-u', 'habitat_baselines/run.py', '--exp-config', 'habitat_baselines/config/objectnav/ddppo_objectnav.yaml', '--run-type', 'train']' returned non-zero exit status 1.```

erikwijmans, 2020-02-21:
I am still not seeing anything about EGL in logs, which is concerning/confusing.  What is your system memory amount?  The ObjectNav dataset currently takes a lot of memory (we are fixing that in PR #309), so maybe you are getting killed by the OOM killer.

shivanshpatel35, 2020-02-21:
I used `watch free -g` to monitor memory. Before I run the code, free memory is 41GB as shown here:
```
                 total        used        free      shared  buff/cache   available
Mem:             62           7          41           0          13          54
Swap:             0           0           0
```
During the entire runtime of the code, free memory doesn't fall below 36GB

erikwijmans, 2020-02-21:
Does the normal habitat-sim example script work? i.e. does `python examples/example.py --scene data/scene_datasets/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb` while in `/path/to/habitat-sim` work?

shivanshpatel35, 2020-02-21:
Yes, the normal habitat-sim example works. Final perfomance printed by running `python examples/example.py --scene data/scene_datasets/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb` is:
```
 ========================= Performance ======================== 
 640 x 480, total time 1.61 s, frame time 1.610 ms (621.2 FPS)
 ==============================================================
```

erikwijmans, 2020-02-21:
Can you run things directly, maybe torch.distributed.launch is blocking the output of something important.  i.e just `python habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ddppo_pointnav.yaml --run-type train`?

shivanshpatel35, 2020-02-21:
Running things without `torch.distributed.launch` results the following log
```
2020-02-21 15:12:22,378 config: BASE_TASK_CONFIG_PATH: configs/tasks/objectnav_mp3d.yaml
CHECKPOINT_FOLDER: new_checkpoints
CHECKPOINT_INTERVAL: 50
CMD_TRAILING_OPTS: []
ENV_NAME: NavRLEnv
EVAL:
  SPLIT: val
  USE_CKPT_CONFIG: True
EVAL_CKPT_PATH_DIR: new_checkpoints
LOG_FILE: train.log
LOG_INTERVAL: 10
NUM_PROCESSES: 1
NUM_UPDATES: 10000
ORBSLAM2:
  ANGLE_TH: 0.2617993877991494
  BETA: 100
  CAMERA_HEIGHT: 1.25
  DEPTH_DENORM: 10.0
  DIST_REACHED_TH: 0.15
  DIST_TO_STOP: 0.05
  D_OBSTACLE_MAX: 4.0
  D_OBSTACLE_MIN: 0.1
  H_OBSTACLE_MAX: 1.25
  H_OBSTACLE_MIN: 0.375
  MAP_CELL_SIZE: 0.1
  MAP_SIZE: 40
  MIN_PTS_IN_OBSTACLE: 320.0
  NEXT_WAYPOINT_TH: 0.5
  NUM_ACTIONS: 3
  PLANNER_MAX_STEPS: 500
  PREPROCESS_MAP: True
  SLAM_SETTINGS_PATH: habitat_baselines/slambased/data/mp3d3_small1k.yaml
  SLAM_VOCAB_PATH: habitat_baselines/slambased/data/ORBvoc.txt
RL:
  DDPPO:
    backbone: resnet50
    distrib_backend: NCCL
    num_recurrent_layers: 2
    pretrained: False
    pretrained_encoder: False
    pretrained_weights: data/ddppo-models/gibson-2plus-resnet50.pth
    reset_critic: True
    rnn_type: LSTM
    sync_frac: 0.6
    train_encoder: True
  PPO:
    clip_param: 0.2
    entropy_coef: 0.01
    eps: 1e-05
    gamma: 0.99
    hidden_size: 512
    lr: 2.5e-06
    max_grad_norm: 0.2
    num_mini_batch: 1
    num_steps: 128
    ppo_epoch: 2
    reward_window_size: 50
    tau: 0.95
    use_gae: True
    use_linear_clip_decay: False
    use_linear_lr_decay: False
    use_normalized_advantage: False
    value_loss_coef: 0.5
  REWARD_MEASURE: distance_to_goal
  SLACK_REWARD: -0.01
  SUCCESS_MEASURE: spl
  SUCCESS_REWARD: 2.5
SENSORS: ['DEPTH_SENSOR', 'RGB_SENSOR']
[... 273 lines cut ...]
    trainer.train()
  File "/local-scratch/habitat-api1/habitat_baselines/rl/ppo/ppo_trainer.py", line 300, in train
    episode_counts,
  File "/local-scratch/habitat-api1/habitat_baselines/rl/ppo/ppo_trainer.py", line 146, in _collect_rollout_step
    outputs = self.envs.step([a[0].item() for a in actions])
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 339, in step
    return self.wait_step()
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 326, in wait_step
    observations.append(read_fn())
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7f4aa0e0d7b8>>
Traceback (most recent call last):
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 468, in __del__
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 347, in close
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 250, in recv
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 375, in _recv
AttributeError: 'NoneType' object has no attribute 'BytesIO'
```

erikwijmans, 2020-02-21:
This is odd.  One last idea for debugging:  Change `habitat.VectorEnv` to `habitat.ThreadedVectorEnv` here: https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/common/env_utils.py#L94 from `

shivanshpatel35, 2020-02-21:
I think the code ran but it lead to another issue, possibly becuase reward is `nan`
```
2020-02-21 15:24:50,497 config: BASE_TASK_CONFIG_PATH: configs/tasks/objectnav_mp3d.yaml
CHECKPOINT_FOLDER: new_checkpoints
CHECKPOINT_INTERVAL: 50
CMD_TRAILING_OPTS: []
ENV_NAME: NavRLEnv
EVAL:
  SPLIT: val
  USE_CKPT_CONFIG: True
EVAL_CKPT_PATH_DIR: new_checkpoints
LOG_FILE: train.log
LOG_INTERVAL: 10
NUM_PROCESSES: 1
NUM_UPDATES: 10000
ORBSLAM2:
  ANGLE_TH: 0.2617993877991494
  BETA: 100
  CAMERA_HEIGHT: 1.25
  DEPTH_DENORM: 10.0
  DIST_REACHED_TH: 0.15
  DIST_TO_STOP: 0.05
  D_OBSTACLE_MAX: 4.0
  D_OBSTACLE_MIN: 0.1
  H_OBSTACLE_MAX: 1.25
  H_OBSTACLE_MIN: 0.375
  MAP_CELL_SIZE: 0.1
  MAP_SIZE: 40
  MIN_PTS_IN_OBSTACLE: 320.0
  NEXT_WAYPOINT_TH: 0.5
  NUM_ACTIONS: 3
  PLANNER_MAX_STEPS: 500
  PREPROCESS_MAP: True
  SLAM_SETTINGS_PATH: habitat_baselines/slambased/data/mp3d3_small1k.yaml
  SLAM_VOCAB_PATH: habitat_baselines/slambased/data/ORBvoc.txt
RL:
  DDPPO:
    backbone: resnet50
    distrib_backend: NCCL
    num_recurrent_layers: 2
    pretrained: False
    pretrained_encoder: False
    pretrained_weights: data/ddppo-models/gibson-2plus-resnet50.pth
    reset_critic: True
    rnn_type: LSTM
    sync_frac: 0.6
    train_encoder: True
  PPO:
    clip_param: 0.2
    entropy_coef: 0.01
    eps: 1e-05
    gamma: 0.99
    hidden_size: 512
    lr: 2.5e-06
    max_grad_norm: 0.2
    num_mini_batch: 1
    num_steps: 128
    ppo_epoch: 2
    reward_window_size: 50
    tau: 0.95
    use_gae: True
    use_linear_clip_decay: False
    use_linear_lr_decay: False
    use_normalized_advantage: False
    value_loss_coef: 0.5
  REWARD_MEASURE: distance_to_goal
  SLACK_REWARD: -0.01
  SUCCESS_MEASURE: spl
  SUCCESS_REWARD: 2.5
SENSORS: ['DEPTH_SENSOR', 'RGB_SENSOR']
[... 273 lines cut ...]
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [3,0,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [4,0,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [5,0,0] Assertion `val >= zero` failed.
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 68, in <module>
    main()
  File "habitat_baselines/run.py", line 38, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 62, in run_exp
    trainer.train()
  File "/local-scratch/habitat-api1/habitat_baselines/rl/ppo/ppo_trainer.py", line 300, in train
    episode_counts,
  File "/local-scratch/habitat-api1/habitat_baselines/rl/ppo/ppo_trainer.py", line 146, in _collect_rollout_step
    outputs = self.envs.step([a[0].item() for a in actions])
  File "/local-scratch/habitat-api1/habitat_baselines/rl/ppo/ppo_trainer.py", line 146, in <listcomp>
    outputs = self.envs.step([a[0].item() for a in actions])
RuntimeError: CUDA error: device-side assert triggered
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.ThreadedVectorEnv object at 0x7fad4edfb940>>
Traceback (most recent call last):
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 468, in __del__
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 350, in close
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/queue.py", line 145, in put
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/threading.py", line 347, in notify
TypeError: 'NoneType' object is not callable
```

shivanshpatel35, 2020-02-21:
Running it with `CUDA_LAUNCH_BLOCKING=1` resulted in an elaborate error. I am copying only the last part of the log.
```2020-02-21 15:44:22,225 update: 10	fps: 41.957	
2020-02-21 15:44:22,225 update: 10	env-time: 23.628s	pth-time: 8.889s	frames: 1408
2020-02-21 15:44:22,225 Average window size 11 reward: nan
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [0,0,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [1,0,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [2,0,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [3,0,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [4,0,0] Assertion `val >= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::<unnamed>::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [5,0,0] Assertion `val >= zero` failed.
THCudaCheck FAIL file=/pytorch/aten/src/THC/generic/THCTensorScatterGather.cu line=67 error=710 : device-side assert triggered
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 68, in <module>
    main()
  File "habitat_baselines/run.py", line 38, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 62, in run_exp
    trainer.train()
  File "/local-scratch/habitat-api1/habitat_baselines/rl/ppo/ppo_trainer.py", line 300, in train
    episode_counts,
  File "/local-scratch/habitat-api1/habitat_baselines/rl/ppo/ppo_trainer.py", line 139, in _collect_rollout_step
    rollouts.masks[rollouts.step],
  File "/local-scratch/habitat-api1/habitat_baselines/rl/ppo/policy.py", line 50, in act
    action_log_probs = distribution.log_probs(action)
  File "/local-scratch/habitat-api1/habitat_baselines/common/utils.py", line 32, in log_probs
    .log_prob(actions.squeeze(-1))
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/site-packages/torch-1.4.0-py3.6-linux-x86_64.egg/torch/distributions/categorical.py", line 116, in log_prob
    return log_pmf.gather(-1, value).squeeze(-1)
RuntimeError: cuda runtime error (710) : device-side assert triggered at /pytorch/aten/src/THC/generic/THCTensorScatterGather.cu:67
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.ThreadedVectorEnv object at 0x7fd5ef808940>>
Traceback (most recent call last):
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 468, in __del__
  File "/local-scratch/habitat-api1/habitat/core/vector_env.py", line 350, in close
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/queue.py", line 145, in put
  File "/local-scratch/anaconda3/envs/habitat/lib/python3.6/threading.py", line 347, in notify
TypeError: 'NoneType' object is not callable

erikwijmans, 2020-02-22:
@mathfac have you seen nan rewards with the objectnav dataset?

erikwijmans, 2020-02-23:
I am not sure why reward is showing up a `nan`, but you are currently using just 1 process, which will make RL highly unstable and wold explain this error.

shivanshpatel35, 2020-03-02:
You are completely right. I changed `LOG_INTERVAL` in `habitat_baselines/config/objectnav/ddppo_objectnav.yaml` to 1 and it runs around 7-8 updates. After that reward turns `nan`.

Thanks for all the help!

closed 2020-03-02 by shivanshpatel35

mathfac, 2020-03-03:
There was no NaN rewards before, but that will definitely break training.
Debugging on my side.

StOnEGiggity, 2020-03-18:
Hi, I find NaN rewards bug may occur when targets are in different large flat(ish) surfaces, which is reported in https://github.com/facebookresearch/habitat-sim/issues/405. I find one example for objectnav: 
```
scene_id: data/scene_datasets/mp3d/oLBMNvg9in8/oLBMNvg9in8.glb
episode_id: 1336
```
You can check it.

srama2512, 2020-07-16:
I'm getting DistanceToGoal metric as NaN in the PointNav training on MP3D. The episode ID might be different than the standard datasets since I made some changes. So, I've given the scene id, agent position and goal position.

```
Episode id:  67230
Scene id: data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.glb
Agent position: [ -1.4486076   -0.35115004 -34.537006  ]
Goal position: [0.5637829899787903, -0.15369552373886108, -42.55187225341797]
```

I haven't had the time to investigate why this is happening. Temporarily, I'm replacing geodesic distance with euclidean distance if this happens so that I can continue to train my agents.

erikwijmans, 2020-07-16:
We fixed a bug in the navmeshes a while back that made a few episodes for MP3D pointnav invalid (they shouldn't have ever been valid and I have no idea how they ever were), this is the script I made for finding/removing them: https://gist.github.com/erikwijmans/e4410f0e12facb87890e919aa264e3fe -- They are just train episodes fortunately

@mathfac did the cleaned up versions ever get re-uploaded?

srama2512, 2020-07-17:
Got it. Thanks!

mathfac, 2020-07-28:
@erikwijmans, thank you for bringing it up. The PointNav v1 MP3D dataset was updated with remove of that faulty episodes: https://dl.fbaipublicfiles.com/habitat/data/datasets/pointnav/mp3d/v1/pointnav_mp3d_v1.zip. cc @srama2512

srama2512, 2020-08-27:
Did the NaN rewards for ObjectNav ever get fixed? I am training DD-PPO agents for ObjectNav and it keeps detecting NaN in the DistanceToGoal metric. For a lot of episodes. Here is a small sample of the episodes where this happens:

```
Episode id: data/scene_datasets/mp3d/b8cTxDM8gDG/b8cTxDM8gDG.glb 16950
Episode id: data/scene_datasets/mp3d/b8cTxDM8gDG/b8cTxDM8gDG.glb 15529
Episode id: data/scene_datasets/mp3d/PX4nDJXEHrG/PX4nDJXEHrG.glb 19029
Episode id: data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.glb 9722
Episode id: data/scene_datasets/mp3d/Uxmj2M2itWa/Uxmj2M2itWa.glb 111
Episode id: data/scene_datasets/mp3d/Uxmj2M2itWa/Uxmj2M2itWa.glb 2174
Episode id: data/scene_datasets/mp3d/Uxmj2M2itWa/Uxmj2M2itWa.glb 813
Episode id: data/scene_datasets/mp3d/Uxmj2M2itWa/Uxmj2M2itWa.glb 2360
Episode id: data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.glb 3237
Episode id: data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.glb 2914
Episode id: data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.glb 11041
```

erikwijmans, 2020-08-27:
I have never be able to reproduce nan/inf distances for ObjectNav.  Did you modify the agent size/height or are you doing something like a teleport agent?

I am running a checker on those scenes to see if anything isn't navigable.

srama2512, 2020-08-27:
Oh, right. I'm using the original agent configuration and not the modified one used for the challenge. Could that be the problem?

mathfac, 2020-08-27:
Yes, that can be a problem.

mathfac, 2020-08-27:
The dataset and spawn points and view points are created for specific agent
specification.

srama2512, 2020-08-27:
Got it. Are there pre-trained DD-PPO agents available for PointNav with the new agent configuration? The mismatch in the observation space and action space (10 -> 30 deg rotations) might be a problem moving forward.