facebookresearch/habitat-lab #311: Errors in training using provided config

saimwani, 2020-02-23:
## ❓ Questions and Help
Hi,
I am trying to reproduce the results on PointNav task (on Matterport3D dataset) using config  [habitat_baselines/config/pointnav/ppo_pointnav.yaml](https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/config/pointnav/ppo_pointnav.yaml). This runs into errors after a few hundred to thousand updates. I tried to print actions during rollout collection and the policy seems to converge to a single action. Can you please help me with this?
```
......
......
......
<pre>2020-02-22 12:18:13,421 update: 100	env-time: 141.647s	pth-time: 68.360s	frames: 38784
2020-02-22 12:18:13,421 Average window size 50 reward: nan
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [1,0,0], thread: [0,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [1,0,0], thread: [1,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [1,0,0], thread: [2,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [1,0,0], thread: [3,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [2,0,0], thread: [0,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [2,0,0], thread: [1,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [2,0,0], thread: [2,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [2,0,0], thread: [3,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [0,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [1,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [2,0,0] Assertion `val &gt;= zero` failed.
/pytorch/aten/src/ATen/native/cuda/MultinomialKernel.cu:243: void at::native::&lt;unnamed&gt;::sampleMultinomialOnce(long *, long, int, scalar_t *, scalar_t *, int, int) [with scalar_t = float, accscalar_t = float]: block: [0,0,0], thread: [3,0,0] Assertion `val &gt;= zero` failed.
Traceback (most recent call last):
  File &quot;habitat_baselines/run.py&quot;, line 218, in &lt;module&gt;
    main()
  File &quot;habitat_baselines/run.py&quot;, line 168, in main
    run_exp(**vars(args))
  File &quot;habitat_baselines/run.py&quot;, line 212, in run_exp
    trainer.train()
  File &quot;/local-scratch/_saim_/habitatAPI/habitat_baselines/rl/ppo/ppo_trainer.py&quot;, line 298, in train
    episode_counts,
  File &quot;/local-scratch/_saim_/habitatAPI/habitat_baselines/rl/ppo/ppo_trainer.py&quot;, line 144, in _collect_rollout_step
    outputs = self.envs.step([a[0].item() for a in actions])
  File &quot;/local-scratch/_saim_/habitatAPI/habitat_baselines/rl/ppo/ppo_trainer.py&quot;, line 144, in &lt;listcomp&gt;
    outputs = self.envs.step([a[0].item() for a in actions])
RuntimeError: CUDA error: device-side assert triggered
Exception ignored in: &lt;bound method VectorEnv.__del__ of &lt;habitat.core.vector_env.VectorEnv object at 0x7f8a3dd36da0&gt;&gt;
Traceback (most recent call last):
  File &quot;/local-scratch/_saim_/habitatAPI/habitat/core/vector_env.py&quot;, line 468, in __del__
  File &quot;/local-scratch/_saim_/habitatAPI/habitat/core/vector_env.py&quot;, line 350, in close
  File &quot;/local-scratch/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py&quot;, line 206, in send
AttributeError: &apos;NoneType&apos; object has no attribute &apos;dumps&apos;</pre>
```
Thanks!

erikwijmans, 2020-02-23:
Looking at your logging, it seems that you set `NUM_PROCESSES` to 3.  This is likely the cause of the instability (RL can be really finicky).  4 processes should fit on a single GPU no problem with the most recent version of the habitat version of MP3D (we did a lot of work on texture compression to make the meshes take way less GPU memory) -- if you haven't downloaded the habitat version of mp3d recently, please re-download it.

saimwani, 2020-02-27:
Thanks for the prompt reply. I have tried training the depth-agent agent using `NUM_PROCESSES: 6` and `num_mini_batch: 4` as mentioned in the paper. I used one GPU to host the model and 3 GPUs for the simulators (2 each). I keep getting the same error after about 5-10 Mn steps. 
With blind agent, the same error appears after less than 100,000 steps. 
This is the config (ppo_pointnav.yaml) I used for depth agent:
```
BASE_TASK_CONFIG_PATH: "configs/tasks/pointnav_mp3d.yaml"
TRAINER_NAME: "ppo"
ENV_NAME: "NavRLEnv"
SIMULATOR_GPU_ID: 0
TORCH_GPU_ID: 1
VIDEO_OPTION: []
# Can be uncommented to generate videos.
# VIDEO_OPTION: ["disk", "tensorboard"]
TENSORBOARD_DIR: "tb_train_depth_20000_6"
VIDEO_DIR: "video_dir"
TEST_EPISODE_COUNT: 994
EVAL_CKPT_PATH_DIR: "cpt/depth_20000_6"
# This was 6 for mp3d and 8 for gibson in the paper
NUM_PROCESSES: 6
# Note:  To train the an RGB only model,
# you may need to use 8 processes with 4 mini batches,
# If so, the number of updates should be cut in half
SENSORS: ["DEPTH_SENSOR"]
CHECKPOINT_FOLDER: "cpt/depth_20000_6"
NUM_UPDATES: 20000
LOG_INTERVAL: 10
CHECKPOINT_INTERVAL: 1000

RL:
  PPO:
    # ppo params
    clip_param: 0.1
    ppo_epoch: 4
    # This was 4 in the paper
    num_mini_batch: 4
    value_loss_coef: 0.5
    entropy_coef: 0.01
    lr: 2.5e-4
    eps: 1e-5
    max_grad_norm: 0.5
    num_steps: 128
    hidden_size: 512
    use_gae: True
    gamma: 0.99
    tau: 0.95
    use_linear_clip_decay: True
    use_linear_lr_decay: True
    reward_window_size: 50
```
This is my `construct_envs` function in `habitat_baselines/common/env_utils.py`:
```
def construct_envs(
    config: Config, env_class: Type[Union[Env, RLEnv]]
) -> VectorEnv:
    r"""Create VectorEnv object with specified config and env class type.
    To allow better performance, dataset are split into small ones for
    each individual env, grouped by scenes.

    Args:
        config: configs that contain num_processes as well as information
        necessary to create individual environments.
        env_class: class type of the envs to be created.

    Returns:
        VectorEnv object created according to specification.
    """

    num_processes = config.NUM_PROCESSES
    configs = []
    env_classes = [env_class for _ in range(num_processes)]
    dataset = make_dataset(config.TASK_CONFIG.DATASET.TYPE)
[... 185 lines cut ...]
  File "/project/6031224/shivp/habitat-sim/_saim_/noMap2/habitat_baselines/rl/ppo/policy.py", line 51, in act
    action_log_probs = distribution.log_probs(action)
  File "/project/6031224/shivp/habitat-sim/_saim_/noMap2/habitat_baselines/common/utils.py", line 40, in log_probs
    .log_prob(actions.squeeze(-1))
  File "/project/6031224/shivp/habitat-sim/python3/lib/python3.7/site-packages/torch-1.4.0-py3.7-linux-x86_64.egg/torch/distributions/categorical.py", line 116, in log_prob
    return log_pmf.gather(-1, value).squeeze(-1)
RuntimeError: cuda runtime error (710) : device-side assert triggered at /pytorch/aten/src/THC/generic/THCTensorScatterGather.cu:67
Exception ignored in: <function VectorEnv.__del__ at 0x2b523d483dd0>
Traceback (most recent call last):
  File "/project/6031224/shivp/habitat-sim/_saim_/noMap2/habitat/core/vector_env.py", line 468, in __del__
    self.close()
  File "/project/6031224/shivp/habitat-sim/_saim_/noMap2/habitat/core/vector_env.py", line 350, in close
    write_fn((CLOSE_COMMAND, None))
  File "/cvmfs/soft.computecanada.ca/easybuild/software/2017/Core/python/3.7.4/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/cvmfs/soft.computecanada.ca/easybuild/software/2017/Core/python/3.7.4/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/cvmfs/soft.computecanada.ca/easybuild/software/2017/Core/python/3.7.4/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
/var/spool/slurmd/job37209460/slurm_script: line 17: 141585 Segmentation fault      (core dumped) CUDA_LAUNCH_BLOCKING=1 python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav.yaml --run-type train

```
I ran with `CUDA_LAUNCH_BLOCKING=1 python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav.yaml --run-type train`.
Also, I have re-downloaded the habitat version of mp3d before running this.

saimwani, 2020-03-14:
Hi,
The code does not run into any stability issues when reproduced with the commit ec9557a, which was used for running the experiments for the paper. It seems the issue has to do with refactoring done after that. It would be great if you could make the behavior of the current version compliant with that of commit ec9557a. 
Thanks!

erikwijmans, 2020-03-16:
Quick thing try:  Move these lines https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/rl/ppo/ppo_trainer.py#L332-L338 to after the agent update here: https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/rl/ppo/ppo_trainer.py#L357

PyTorch ~recently changed the way the LR schedulers work (the optimizer now needs to be stepped first), does that fix it?

erikwijmans, 2020-03-16:
Took a closer look.  There a select few episodes that are now invalid after fixing a bug that seemingly should have had no effect on episode navigability.  Fortunately, this only happened in train, so it is straightforward to just remove them: https://gist.github.com/erikwijmans/e4410f0e12facb87890e919aa264e3fe

closed 2020-03-29 by saimwani