facebookresearch/habitat-lab #313: Windows 10 support?

quark2019, 2020-02-24:
The habitat docker requires  nvidia-docker, however,  nvidia-docker doesnot support Windows.
(The offical team of nvidia-docker has not any plan to support Windows.)

1. Is there any way to run habitat docker on Windows 10 ?
2. Does habitat support Windows 10 (even if build from source codes) ?

(I'm using Windows 10 Pro, and Pytorch with CUDA works fine for me.)

Thanks！

erikwijmans, 2020-02-24:
Nvidia docker is required to run habitat, so no, there is no way to run the docker on windows.

There isn't official compatibility for building from source on windows, but some community members have been working on that here https://github.com/facebookresearch/habitat-sim/pull/84

rpartsey, 2022-09-01:
Stale.

closed 2022-09-01 by rpartsey