facebookresearch/habitat-lab #315: How to accuess all reachable positions within a Gibson dataset for a habitat agent?

AAdeleye, 2020-02-26:
## ❓ Questions and Help
Hello,

This question was addressed partially [here](https://github.com/facebookresearch/habitat-api/issues/261#issue-530931396) but I had a follow up question. 

Is it possible to access a list of reachable positions within a scene. 
For instance, if I load "Andover", rather than move randomly through the environment to generate images that I can save, I would want to loop over a list of coordinates. That way I have some confidence that images of the environment are at every room. I know the .jason.gz files contain random start positions. Would this be the best way to achieve this?

Also, for the files in the PointNav dataset for Gibsion. I was wondering if the .jason.gz files start positions are all on the first floor? (for a robot that can't go up stairs)  

Thank you.

erikwijmans, 2020-02-26:
Relevant: https://github.com/facebookresearch/habitat-sim/pull/483

There isn't a list of reachable locations, the navigation space is infinite.  The PR linked above does a uniform sampling of locations over a floor.

AAdeleye, 2020-03-31:
Thank you, that worked well. 

I know the code is still being updated but A few of the scenes, at least in the Gibson Dataset, have more distortion that I thought. 
[image]
[image]

It doesn't seem to be impact my network too much but I was wondering if this is to be expected?
Thanks -I will close after this since it's diverging a bit.

dhruvbatra, 2020-03-31:
Yes, that is the data. Habitat doesn't create the data, we just render/simulate it. You can read about our quality annotations in our ICCV19 paper.

closed 2020-03-31 by AAdeleye