facebookresearch/habitat-lab #321: Use Heading sensor and Top Down Map without a task

tfy14esa, 2020-03-04:
## ❓ Questions and Help
Hi,
I have written code to explore an arbitrary scene with the keyboard. This is done similar to the recipe described [here](https://aihabitat.org/docs/habitat-sim/stereo-agent.html), i.e. I only use habitat_sim and I do not load a config yaml file to initialize an environment. Instead I configure a habitat simulator according to the stereo agent code as follows:
`sim = habitat_sim.Simulator(habitat_sim.Configuration(backend_cfg, [agent_config]))`

I would like to add a top down map of the scene which is to be updated as I traverse the scene. In the example script "shortest_path_follower_example.py", I find the necessary code, which creates an environment with a "HEADING"-sensor and a "TOP_DOWN_MAP"-measurement.  

I would like to add a HEADING-sensor and a TOP_DOWN_MAP-measurement to my simulator. According to the [documentation](https://tmp.mosra.cz/habitat-api/habitat.core.simulator.html#SensorTypes-HEADING), it is possible to define the HEADING sensor in habitat-api, but according to this [document](https://tmp.mosra.cz/habitat-sim/habitat_sim.sensor.html#SensorType), it is not possible with habitat_sim. 

Is there any way to get a top down map without having to define a task as is done in "shortest_path_follower_example.py"?

Thank you,

erikwijmans, 2020-03-04:
Short answer: No, the sensors and measure in habitat-api (this repo) are designed to work the environments.

It may be worth looking at this PR: https://github.com/facebookresearch/habitat-sim/pull/483 and the other ones from the same author.

tfy14esa, 2020-03-05:
Ok thank you!

closed 2020-03-05 by erikwijmans