facebookresearch/habitat-lab #322: To train DD-PPO (depth model) using Gibson 4+ and MP3D (train/val/test)

Shigangli, 2020-03-04:
## ❓ Questions and Help
[image]

erikwijmans, 2020-03-05:
Hi, 

1) I used a variant of this script: https://gist.github.com/erikwijmans/1c596183fb8de1e9afd80088b6b5c115 to generate the pointnav episodes (generated 50k per scene).

2) All val and test scores are reported on Gibson, so use the Gibson val and test scenes, https://github.com/facebookresearch/habitat-api#task-datasets and https://evalai.cloudcv.org/web/challenges/challenge-page/254/leaderboard/837

3) The only thing you need to change in one of the base pointnav configs is the DATA_PATH (i.e. here: https://github.com/facebookresearch/habitat-api/blob/master/configs/tasks/pointnav_gibson.yaml#L31  The only changed needed for the https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/config/pointnav/ddppo_pointnav.yaml file is to point to the new task config.

erikwijmans, 2020-03-06:
The number of machines shouldn't effect the amount of GPU memory used.  DD-PPO uses PyTorch's DistributedDataParallel to accumulate gradients across different GPUs, so you may have better luck debugging that by tasking there.

To lower memory usage, you can change the parameter here: https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/config/pointnav/ddppo_pointnav.yaml#L30 (less steps, less memory usage, shouldn't hurt final performance too much).  One other option is to use resnet18 instead of resnet50 -- for depth at least, my experience is that the larger CNNs do provide additional performance, but it is a small gain.

Shigangli, 2020-03-11:
I can train DD-PPO using more machines now. I trained the model for 100 updates for a small scale test.

But I didn't see the SPL scores printed out. I also visualize the tensorboard files but there is only the curve for the reward value. The command I used is as follows.

<srun python -u -m habitat_baselines.run --exp-config habitat_baselines/config/pointnav/ddppo_pointnav.yaml --run-type train>

How can I generate the curve for the SPL scores?

erikwijmans, 2020-03-12:
See https://github.com/facebookresearch/habitat-api/pull/332 for logging of things

For eval, yes, use a 1 GPU machine (the evaluation only runs on 1 GPU).  You'll want to use `--run-type eval` for eval.

Shigangli, 2020-03-14:
Thanks. I made a typo :-)

closed 2022-09-01 by rpartsey