facebookresearch/habitat-lab #325: How could DDPPO run on multiple GPUs?

qianLyu, 2020-03-08:
Hi all,

I tried to run DDPPO on multiple GPUs with the code:
`python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ddppo_pointnav.yaml --run-type train`
and the config:
```
SIMULATOR_GPU_ID: 0
TORCH_GPU_ID: 1
```

However, it seems like GPU1 is not working(I have 4 GPUs in total):
```
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Tesla P40           Off  | 00000035:00:00.0 Off |                    0 |
| N/A   37C    P0   128W / 250W |   4372MiB / 22919MiB |     83%      Default |
+-------------------------------+----------------------+----------------------+
|   1  Tesla P40           Off  | 00006F65:00:00.0 Off |                    0 |
| N/A   25C    P8    12W / 250W |     10MiB / 22919MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   2  Tesla P40           Off  | 0000736F:00:00.0 Off |                    0 |
| N/A   27C    P8    12W / 250W |     10MiB / 22919MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   3  Tesla P40           Off  | 0000BD86:00:00.0 Off |                    0 |
| N/A   23C    P8    11W / 250W |     10MiB / 22919MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0     55411      C   python                                      3737MiB |
|    0     55457      G   /home/qian/anaconda3/bin/python              256MiB |
|    0     55458      G   /home/qian/anaconda3/bin/python              375MiB |
+-----------------------------------------------------------------------------+
```

The GPU1 is working when I run PPO baseline, so what's the problem with DDPPO?

Another question: How could I use all the four GPUs in one training process? It seems that I can only choose one for SIMULATOR_GPU and the other for TORCH_GPU.

Thanks a lot!

erikwijmans, 2020-03-08:
Please use one of the two provided scripts for launching multiple workers to use multiple GPUs.  Note that DD-PPO ignores `SIMULATOR_GPU_ID` and `TORCH_GPU_ID` entirely and assigns GPUs sequentially to workers (worker 0 uses GPU 0, etc).

https://github.com/facebookresearch/habitat-api/tree/master/habitat_baselines/rl/ddppo#running

closed 2020-03-08 by erikwijmans

qianLyu, 2020-03-09:
Thank you! I could now run DD-PPO on multiple GPUs

bareblackfoot, 2020-05-30:
@erikwijmans I tried your solution but it does not work on my case :(
When I run multi_node_slurm.sh, it only uses a single GPU and show a OOM error. How can I use multi-GPUs? 

-----
error log:
2020-05-31 01:56:22,368 Initializing dataset PointNav-v1
2020-05-31 01:56:24,598 Initializing dataset PointNav-v1
2020-05-31 01:56:24,754 Initializing dataset PointNav-v1
2020-05-31 01:56:24,776 Initializing dataset PointNav-v1
2020-05-31 01:56:24,810 Initializing dataset PointNav-v1
2020-05-31 01:56:57,279 initializing sim Sim-v0
2020-05-31 01:56:59,528 initializing sim Sim-v0
2020-05-31 01:56:59,725 initializing sim Sim-v0
2020-05-31 01:57:01,897 initializing sim Sim-v0
2020-05-31 01:57:18,310 Initializing task Nav-v0
2020-05-31 01:57:18,665 Initializing task Nav-v0
2020-05-31 01:57:58,273 Initializing task Nav-v0
2020-05-31 01:58:17,207 Initializing task Nav-v0
2020-05-31 01:58:17,566 Overwriting CNN input size of depth: (256, 256)
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/blackfoot/programs/habitat-api/habitat_baselines/run.py", line 70, in <module>
    main()
  File "/home/blackfoot/programs/habitat-api/habitat_baselines/run.py", line 39, in main
    run_exp(**vars(args))
  File "/home/blackfoot/programs/habitat-api/habitat_baselines/run.py", line 64, in run_exp
    trainer.train()
  File "/home/blackfoot/programs/habitat-api/habitat_baselines/rl/ddppo/algo/ddppo_trainer.py", line 181, in train
    self._setup_actor_critic_agent(ppo_cfg)
  File "/home/blackfoot/programs/habitat-api/habitat_baselines/rl/ddppo/algo/ddppo_trainer.py", line 82, in _setup_actor_critic_agent
    self.actor_critic.to(self.device)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 426, in to
    return self._apply(convert)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 202, in _apply
    module._apply(fn)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 202, in _apply
    module._apply(fn)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 224, in _apply
    param_applied = fn(param)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 424, in convert
    return t.to(device, dtype if t.is_floating_point() else None, non_blocking)
RuntimeError: CUDA error: out of memory
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7ff8cbae37b8>>
Traceback (most recent call last):
  File "/home/blackfoot/programs/habitat-api/habitat/core/vector_env.py", line 518, in __del__
    self.close()
  File "/home/blackfoot/programs/habitat-api/habitat/core/vector_env.py", line 400, in close
    write_fn((CLOSE_COMMAND, None))
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
srun: error: eevee: task 0: Exited with exit code 1

erikwijmans, 2020-05-30:
Hi, can you please provide more information.  How did you run `multi_node_slurm.sh` (FYI, that script _only_ makes sense to use on a SLURM cluster)?  It seems like you are OOMing on setup, which indicates either something else is running on the GPU, or its amount of memory is small (i.e. less than 6GB).

bareblackfoot, 2020-05-30:
Thank you for the fast answer :)
I run the bash code in a terminal.
blackfoot@eevee:~/programs/habitat-api$ bash ./habitat_baselines/rl/ddppo/multi_node_slurm.sh

The memory is over 12GB
[image]

erikwijmans, 2020-05-30:
You may need to reduce the rollout length (`num_steps` in the config under PPO) (to say 96 or 64) to get ResNet50 to fit on Titan XP.

As far as only running on 1 GPU, how did you get the slurm reservation?  The slurm script either needs to be run as a batch script (via `sbatch`) or requires an `salloc` reservation, i.e.
```
salloc --gres gpu:4 --ntasks-per-node 4 --nodes 1 bash
```
For 4 GPUs 1 node.  You should double check with your cluster admins however, there are different ways to setup SLURM that may cause this to not work.

bareblackfoot, 2020-05-31:
Since I do not know how to use slurm, I just changed to run single_node.sh.
When I change the num_steps very small (tried 64, 32...) But It also showed OOM.
I am using a master branch, not a stable branch to use updated code. Maybe it could be a problem..?

bareblackfoot, 2020-05-31:
I changed the server which has a 24GB GPU and now it works.. :)

erikwijmans, 2020-05-31:
Ah, for mp3d please make sure you have downloaded the habitat-version of the dataset in the past few months.  We recently re-released it with texture compressed meshes that use a lot less GPU memory.

bareblackfoot, 2020-06-01:
You mean, downloading mp3d again by a script from "https://niessner.github.io/Matterport/"?
Thanks a lot!

erikwijmans, 2020-06-01:
Yep