facebookresearch/habitat-lab #329: How to align depth sensor and semantic sensor?

lynl7130, 2020-03-11:
## How to align depth sensor and semantic sensor?

Hi, my task request sophisticated alignment between semantic sensor output and depth sensor output. However, currently habitat's two sensors are not aligned. 
The image attached is an example:
[image]

Is there a way for me to get aligned outputs of these sensors? Thanks in advance!

lynl7130, 2020-03-21:
Changing the rendering mesh [here](https://github.com/facebookresearch/habitat-sim/blob/171793312525155084393f0ea16c668ea354a6da/habitat_sim/simulator.py#L399-L406) solved my problem!

closed 2020-03-21 by lynl7130

ethanabrooks, 2022-03-21:
@lynl7130 could you say what you changed exactly?