facebookresearch/habitat-lab #330: Can't visualise the Example

chale96, 2020-03-11:
## ❓ Questions and Help

I am trying to run the example from [habitat-api's quick start guide](https://github.com/facebookresearch/habitat-api/blob/master/docs/pages/quickstart.rst) but when I run it, it doesn't open the viewer. 

I create the file `testActions.py` and I am calling it from outside both habitat-api and habitat-sim repository. I tried bot `python testActions.py` and `python testActions.py ./habitat-sim/build/viewer --enable-physics /data/scene_datasets/habitat-test-scenes/van-gogh-room.glb`

this is what it returns:

```
2020-03-11 15:43:12,344 Initializing dataset PointNav-v1
2020-03-11 15:43:12,562 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0311 15:43:12.599169 233229760 SceneGraph.h:92] Created DrawableGroup: 
Renderer: Intel(R) Iris(TM) Plus Graphics 640 by Intel Inc.
OpenGL version: 4.1 INTEL-14.4.23
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_vertex_array_object
    GL_EXT_debug_label
    GL_EXT_debug_marker
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
I0311 15:43:12.769114 233229760 ResourceManager.cpp:1072] Importing Basis files as BC3
W0311 15:43:14.076529 233229760 Simulator.cpp:140] :
---
 The active scene does not contain semantic annotations. 
---
I0311 15:43:14.099301 6910 simulator.py:142] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
2020-03-11 15:43:14,156 Initializing task Nav-v0
Environment creation successful
Traceback (most recent call last):
  File "testActions.py", line 65, in <module>
    example()
  File "testActions.py", line 24, in example
    observations["pointgoal"][0], observations["pointgoal"][1]))
KeyError: 'pointgoal'
I0311 15:43:14.704975 233229760 Simulator.cpp:35] Deconstructing Simulator
I0311 15:43:14.705004 233229760 SemanticScene.h:40] Deconstructing SemanticScene
I0311 15:43:14.705010 233229760 SceneManager.h:24] Deconstructing SceneManager
I0311 15:43:14.705014 233229760 SceneGraph.h:25] Deconstructing SceneGraph
I0311 15:43:14.705575 233229760 RenderTarget.h:51] Deconstructing RenderTarget
I0311 15:43:14.705756 233229760 Sensor.h:80] Deconstructing Sensor
I0311 15:43:14.715387 233229760 Renderer.cpp:33] Deconstructing Renderer
I0311 15:43:14.715456 233229760 WindowlessContext.h:16] Deconstructing WindowlessContext
I0311 15:43:14.715466 233229760 WindowlessContext.cpp:264] Deconstructing GL context
```

dhruvbatra, 2020-03-11:
@abhiskk

chale96, 2020-03-12:
Am I running it with the wrong commands?

chale96, 2020-03-13:
@mosra  @abhiskk

erikwijmans, 2020-03-13:
Looks like that example hasn't been updated since we renamed sensors.  Change `pointgoal` to `pointgoal_with_gps_compass`.  The first variant of the command, `python testActions.py` is correct (`build/viewer` is not usable from python, it is a stand-alone program).

chale96, 2020-03-13:
> Looks like that example hasn't been updated since we renamed sensors. Change `pointgoal` to `pointgoal_with_gps_compass`. The first variant of the command, `python testActions.py` is correct (`build/viewer` is not usable from python, it is a stand-alone program).

Thank you!

now it is giving me `qt.qpa.plugin: Could not find the Qt platform plugin "cocoa" in "" ` error.

```
2020-03-13 17:29:37,024 Initializing dataset PointNav-v1
2020-03-13 17:29:37,217 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0313 17:29:37.234107 303062464 SceneGraph.h:92] Created DrawableGroup: 
Renderer: Intel(R) Iris(TM) Plus Graphics 640 by Intel Inc.
OpenGL version: 4.1 INTEL-14.4.23
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_vertex_array_object
    GL_EXT_debug_label
    GL_EXT_debug_marker
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
I0313 17:29:37.395491 303062464 ResourceManager.cpp:1072] Importing Basis files as BC3
W0313 17:29:38.644949 303062464 Simulator.cpp:140] :
---
 The active scene does not contain semantic annotations. 
---
I0313 17:29:38.651930 10376 simulator.py:142] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
2020-03-13 17:29:38,737 Initializing task Nav-v0
Environment creation successful
Destination, distance: 5.883259, theta(radians): -0.00
qt.qpa.plugin: Could not find the Qt platform plugin "cocoa" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

zsh: abort      python testActions.py
```

erikwijmans, 2020-03-13:
That doesn't sound like an issue with habitat, but an issue with your OpenCV install (this little example uses opencv to show images).  I haven't seen that issue before, so I don't know where to start with fixing it.

chale96, 2020-03-13:
There is a problem with the updated version of `opencv-python`. I find the solution by installing the version 4.1.0.25:

```
pip unistall opencv-python
pip install opencv-python==4.1.0.25
```

PR #334 (2020-03-16): pointgoal -> pointgoal_with_gps_compas
## Motivation and Context

Simple change of names.

Closes #330 

## How Has This Been Tested

It works

## Types of changes

- Docs change / refactoring / dependency upgrade
closed 2020-03-13 by chale96