facebookresearch/habitat-lab #337: Not the same coordinate system?

xiangyuanhang, 2020-03-20:
## ❓ Questions and Help
Hi,
I'm trying to select some start/end points manully from Gibson-dataset.
First I opened the glb model using blender.
And then I visualized several points  using the coords in the point-nav json file.
However, these points appear not in the house. The picture below shows the model Adrian.glb and several points from  pointnav/train/content/Adrian.json
[image]

So I wonder the points in the json file  are not in the same coordinate system with the glb model?
If so, how could I get the correct point coords for use in json by choosing manually from the scene?

Thank you so much.

erikwijmans, 2020-03-20:
Yes.  gibson (and mp3d) are -Z gravity meshes.  Habitat rotates these to -Y.  See here: https://github.com/facebookresearch/habitat-sim/issues/524

xiangyuanhang, 2020-03-22:
Thanks for replying!
I have tried the method you provided. Picture below shows what I have done. The mesh is Adrian from Gibson. Basically I choose a goal point from Adrian.json(pointnav) and then I transform the position of the point using the method you mentioned. Then I visualize the point in blender. However, the point is still outside of the house...
[image]
[image]
Is it possible that the method you provide is only for mp3d but not for Gibson?
Thank you so much.

erikwijmans, 2020-03-22:
No, it's the same for Gibson.  IIRC, there is also something weird Blender does with the coordinate system.  CC @aclegg3