facebookresearch/habitat-lab #341: How to use Collisions method

chale96, 2020-03-24:
## ❓ Questions and Help

Are there any examples on how can I call the `Collisions()` method in my file?
The idea is to receive an output whenever a collision occurs.

I currently imported using `from habitat.tasks.nav.nav import Collisions` and then trying to call it in a `print()` but I am getting `TypeError: __init__() missing 2 required positional arguments: 'sim' and 'config'` 

An example should be sufficient to understand how to use it.

erikwijmans, 2020-03-24:
That class isn't designed to work on its own.  Like with other measures, you add it to the list of `MEASURES` in a config (add `'COLLISIONS'` for this one).  It will then show up with the outputs from other measures (like success and SPL) -- generally in the `info` dict if using the gym compatible RLEnv.

chale96, 2020-03-24:
Is there a way to detect collisions?

erikwijmans, 2020-03-24:
You can use the same logic to make a sensor or measure that is true or false if the last action resulted in a collision (the general delineation between sensor vs. measure is that sensors are for inputs to the agent while measures are used to evaluate the agent's performance).  This example shows how to add new sensors/measure: https://github.com/facebookresearch/habitat-api/blob/master/examples/register_new_sensors_and_measures.py

closed 2020-03-24 by chale96