facebookresearch/habitat-lab #346: DD-PPO does not all reduce gradients

Purpleslz, 2020-03-28:
## 🐛 Bug

DD-PPO does not all reduce gradients during the backward call, because `reducer.prepare_for_backward` is not being called during training process.

The problem is in this line: https://github.com/facebookresearch/habitat-api/blob/v0.1.4/habitat_baselines/rl/ddppo/algo/ddppo.py#L96

```
class DecentralizedDistributedMixin:

    ...
    def before_backward(self, loss):
         # ...

# Mixin goes second that way the PPO __init__ will still be called
class DDPPO(PPO, DecentralizedDistributedMixin):  
    # Here PPO and Mixin both have "before_backward" method, 
    # DDPPO will call PPO's not the Mixin's.
    pass
```

And here is a quick fix:
```
class DecentralizedDistributedMixin:
    ...

# Mixin goes second that way the PPO __init__ will still be called
class DDPPO(PPO, DecentralizedDistributedMixin):  

      # Move before_backward to DDPPO
      def before_backward(self, loss):
         # ...
```

erikwijmans, 2020-03-28:
It's actually easier to fix than that, I don't know what happened, but it should `class DDPPO(DecentralizedDistributedMixin, PPO)`.

closed 2020-03-28 by erikwijmans

Xiaoming-Zhao, 2020-04-30:
@erikwijmans Hi, may I know whether the main aim of defining `DecentralizedDistributedMixin` class is for manually adding `loss` into the `Reducer` of PyTorch's `DecentralizedDistributedMixin`? Because the agent in DDPPO does not utilize `forward` pass, I think the [automatically processing](https://github.com/pytorch/pytorch/blob/v1.2.0/torch/nn/parallel/distributed.py#L435) here will not work.

erikwijmans, 2020-04-30:
Yes, that is correct, since the forward pass in not used by the agents, we need to manually make the call to `reducer.prepare_for_backward`.

PR #347 (2020-03-28): Fix gradient reduction
## Motivation and Context

The Mixin order in DD-PPO is wrong.  I have no idea how this escaped my testing when creating the public release, but it did.

Closes #346 

## How Has This Been Tested

Re-checked, `self.reducer.prepare_for_backward` now gets called and re-verified that gradients are the same across all ranks.

## Types of changes

- Bug fix (non-breaking change which fixes an issue)