facebookresearch/habitat-lab #350: How to train with just a custom sensor as input ?

ronjamoller, 2020-03-31:
Hi, i want to train a network that only takes my custom sensor as input - do i have to modify the .yaml file or do i have to manually change the observation space inside the code ? I have tried just removing the sensors from the simulation setup but as expected this will cause an error, i am hesitant to manually change VectorEnv parameters since the whole initialization process is so automated.
Related question: my understanding is that the observation space is automatically generated from the simulation and task sensors, is that correct ? 
thank you,
ronja

erikwijmans, 2020-03-31:
No need to change VectorEnv, the observation space will be automatically generated based on what sensors are active.  You will need to change the .yaml file to instantiate the sensor (or the config object in code).  https://github.com/facebookresearch/habitat-api/blob/master/examples/register_new_sensors_and_measures.py may be helpful.

ronjamoller, 2020-03-31:
Thank you for getting back to me so quickly, my sensor is registered and tested in a test environment (like in the example you linked), the point i am having trouble with is training on just that sensor. I can use it in addtion to simulator sensors like rgb or depth ( observation_space.spaces in PointNavBaselineNet returns OrderedDict([('custom_sensor', Box(32, 2)), ('pointgoal_with_gps_compass', Box(2,)), ('rgb', Box(256, 256, 3))]) and this trains just fine) but when i give SENSORS: [] in the yaml so the observation space is like this: 
OrderedDict([('custom_sensor', Box(32, 2)), ('pointgoal_with_gps_compass', Box(2,))]) 

i get this error (everything but the yaml is unchanged):

F0331 22:33:04.746765  9944 Simulator.cpp:207] Check failed: activeSemanticSceneID_ >= 0 (-1 vs. 0) and no training happens.
Furthermore, i am unsure how to realise my plan of training for a "pointnav" task with just my custom sensor and without feeding the network the pointgoal with gps sensor ( so no goal_sensor_uuid), looking at PointNavBaselineNet my understanding is that i would have to implement my own net as the definition of pointnav here demands a gps input - so just removing the gps sensor like the rgb and depth sensor wont work.
I hope that made sense - the tl;dr version is, i want to train an agent to navigate to a certain point based on just my custom sensor inputs and im having trouble setting it up, using my custom sensor in addition to other sensors works just fine.
thank you

erikwijmans, 2020-03-31:
You will need to change the network to do what you want, habitat-baselines is not intended to encapsulate every possible situation.

The `activeSemanticSceneID_ >= 0` check failing happens when the simulator wasn't configured with any sensors (i.e. no rgb, no depth).  You shouldn't run into that unless you add some sensor afterwards and then try to get observations.  I tried to reproduce this, set `SENSOR: []` in https://github.com/facebookresearch/habitat-api/blob/master/habitat_baselines/config/pointnav/ppo_pointnav_example.yaml, but could not.

ronjamoller, 2020-04-01:
Thank you for confirming that i will need to implement the network myself, it should not be too difficult since i have lots of inspiration to draw from with the existing models. 
As for the activeSemanticSceneID issue - i think it is exactly as you described - i am not initializing the simulation with rgb or depth but i do add the custom sensor before i call the trainer initialization so i assumed the sim would be configured with the new sensor. If i just remove rgb and depth like you said i have no issues, the problem only shows up when i add my custom sensor on top of that. I will look further into the initialization process in case i missed something there.

mathfac, 2020-04-02:
Maybe, what @erikwijmans said is to make sure that your task config and baseline config both have `SENSORS: []` as well as code isn't adding sensors after simulator was initialized.

ronjamoller, 2020-04-02:
Thank you both for taking the time to reply to this - my sensor is added just after the config is loaded.
i have downloaded a fresh version of habitat-api and am running baselines/run, to avoid issues with my custom sensor i am using the register_new_sensors_and_measures.py code for adding a new sensor.

my observation are as follows: 
**fresh download of habitat sim and api, branch stable**

normal baselines/run with ppo_pointnav_example.yaml and pointnav.yaml unchanged, no custom sensor - training and testing works (eval complaints about float in mismatch in habitat/config, can be fixed by setting default params to int)

normal baselines/run with ppo_pointnav_example.yaml and pointnav.yaml both with SENSORS: [], no custom sensor  - can train but eval results in size mismatch for `actor_critic.net.state_encoder.rnn.weight_ih_l0: copying a param with shape torch.Size([1536, 514]) from checkpoint, the shape in current model is torch.Size([1536, 2])`

normal baselines/run with ppo_pointnav_example.yaml and pointnav.yaml both with only ppo_pointnav_example.yaml SENSORS: [] - can train but eval results in size mismatch `size mismatch for actor_critic.net.state_encoder.rnn.weight_ih_l0: copying a param with shape torch.Size([1536, 514]) from checkpoint, the shape in current model is torch.Size([1536, 2])`

normal baselines/run with ppo_pointnav_example.yaml and pointnav.yaml unchanged, with custom sensor can train and eval

normal baselines/run with ppo_pointnav_example.yaml and pointnav.yaml both with SENSORS: [], with custom sensor - can train, fails eval with size mismatch for `actor_critic.net.state_encoder.rnn.weight_ih_l0: copying a param with shape torch.Size([1536, 514]) from checkpoint, the shape in current model is torch.Size([1536, 2]).`

**for fun i tried to just checkout the newest master for habitat-api, habitat-sim is branch stable:**
normal baselines/run with ppo_pointnav_example.yaml and pointnav.yaml unchanged,  no added sensor - training works, first fails with int float mismatch , after that is changed eval fails with `AttributeError: 'PPOTrainer' object has no attribute 'metric_uuid'`

this is my sensor insertion code in baselines/run.py

```

#!/usr/bin/env python3

# Copyright (c) Facebook, Inc. and its affiliates.
# This source code is licensed under the MIT license found in the
# LICENSE file in the root directory of this source tree.

import argparse
import random

import numpy as np

from habitat_baselines.common.baseline_registry import baseline_registry
from habitat_baselines.config.default import get_config
from typing import Any
from gym import spaces
import habitat

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument(
        "--run-type",
        choices=["train", "eval"],
        required=True,
        help="run type of the experiment (train or eval)",
    )
    parser.add_argument(
        "--exp-config",
        type=str,
        required=True,
        help="path to config yaml containing info about experiment",
    )
    parser.add_argument(
        "opts",
        default=None,
        nargs=argparse.REMAINDER,
        help="Modify config options from command line",
    )

    args = parser.parse_args()
    run_exp(**vars(args))

# Define the sensor and register it with habitat
# For the sensor, we will register it with a custom name
@habitat.registry.register_sensor(name="my_supercool_sensor")
class AgentPositionSensor(habitat.Sensor):
    def __init__(self, sim, config, **kwargs: Any):
        super().__init__(config=config)
[... 45 lines cut ...]
    # Use the custom name
    config.TASK_CONFIG.TASK.AGENT_POSITION_SENSOR.TYPE = "my_supercool_sensor"
    config.TASK_CONFIG.TASK.AGENT_POSITION_SENSOR.ANSWER_TO_LIFE = 42
    # Add the sensor to the list of sensors in use
    config.TASK_CONFIG.TASK.SENSORS.append("AGENT_POSITION_SENSOR")
    config.freeze()

    random.seed(config.TASK_CONFIG.SEED)
    np.random.seed(config.TASK_CONFIG.SEED)

    trainer_init = baseline_registry.get_trainer(config.TRAINER_NAME)
    assert trainer_init is not None, f"{config.TRAINER_NAME} is not supported"
    trainer = trainer_init(config)

    if run_type == "train":
        trainer.train()
    elif run_type == "eval":
        trainer.eval()

if __name__ == "__main__":
    main()
```

I have not observed the Scene id error in all of this so my guess is that it was either a version update or my own fault that lead to that issue. Unfortunately this leaves me with the situation that i cannot train and eval on only my sensor (i would have to remove the `POINTGOAL_WITH_GPS_COMPASS_SENSOR` separately but im not at that point yet). 
I am sorry for making this such an issue, but i dont know how to continue at this point.

erikwijmans, 2020-04-02:
>  (eval complaints about float in mismatch in habitat/config, can be fixed by setting default params to int)

Where is that coming from?

> Size mismatch

Looks like you are picking up a checkpoint that had vision sensors, make sure you are pointing eval to the correct checkpoints and/or that there aren't old checkpoints in the folder (if you are pointing eval to a folder).

> `AttributeError: 'PPOTrainer' object has no attribute 'metric_uuid'`

Fixed in #357

ronjamoller, 2020-04-02:
re: question number 1: `ValueError: Type mismatch (<class 'float'> vs. <class 'int'>) with values (0.0 vs. 0) for config key: TASK_CONFIG.SIMULATOR.DEPTH_SENSOR.MIN_DEPTH` came from habitat/config/default.py , if i set min and max depth to int in default.py its fine
Thank you for the tip to delete the checkpoints, I assumed it would automatically take the newest checkpoint and i always ran a training with rgb and depth at the start to double check if everything works. After deleting everything in the checkpoint folder, running a new training with SENSORS: [] in both .yaml files i can train, eval now works (of course only with deactivated video options).
Also thank you for pointing me to the fix with the current master.
All in all i guess this solves my issues - thank you very much, i will now try to find a way to train without the gps sensor.

closed 2020-04-08 by ronjamoller

W-xf, 2020-08-01:
Hello, the error I encountered is:

ValuEerror: type mismatch (< class' habitat.config.default .Config'> vs. <class ' yacs.config.CfgNode '>) with values (DATASET:

I don't understand why the report is wrong. I look forward to your reply.

erikwijmans, 2020-08-01:
Can you please open a new issue as that seems unrelated to this issue. Also, please provide additional details and context,  i.e. how do you reproduce that error?