facebookresearch/habitat-lab #354: Error when running examples/example.py

wuxx1624, 2020-04-01:
I installed the habitat-api and habitat-sim. I got the error  when I tried to run the **python examples/example.py** in habitat-api 

Traceback (most recent call last):
  File "examples/example.py", line 7, in <module>
    import habitat
  File "/home/fei/Documents/habitat-api/habitat/__init__.py", line 9, in <module>
    from habitat.core.benchmark import Benchmark
  File "/home/fei/Documents/habitat-api/habitat/core/benchmark.py", line 17, in <module>
    from habitat.core.env import Env
  File "/home/fei/Documents/habitat-api/habitat/core/env.py", line 18, in <module>
    from habitat.datasets import make_dataset
  File "/home/fei/Documents/habitat-api/habitat/datasets/__init__.py", line 7, in <module>
    from habitat.datasets.registration import make_dataset
  File "/home/fei/Documents/habitat-api/habitat/datasets/registration.py", line 23, in <module>
    _try_register_objectnavdatasetv1()
  File "/home/fei/Documents/habitat-api/habitat/datasets/object_nav/__init__.py", line 20, in _try_register_objectnavdatasetv1
    from habitat.datasets.object_nav.object_nav_dataset import (
  File "/home/fei/Documents/habitat-api/habitat/datasets/object_nav/object_nav_dataset.py", line 15, in <module>
    from habitat.datasets.pointnav.pointnav_dataset import (
  File "/home/fei/Documents/habitat-api/habitat/datasets/pointnav/pointnav_dataset.py", line 15, in <module>
    from habitat.tasks.nav.nav import (
  File "/home/fei/Documents/habitat-api/habitat/tasks/__init__.py", line 7, in <module>
    from habitat.tasks.registration import make_task
  File "/home/fei/Documents/habitat-api/habitat/tasks/registration.py", line 24, in <module>
    _try_register_eqa_task()
  File "/home/fei/Documents/habitat-api/habitat/tasks/eqa/__init__.py", line 13, in _try_register_eqa_task
    from habitat.tasks.eqa.eqa import EQATask
  File "/home/fei/Documents/habitat-api/habitat/tasks/eqa/eqa.py", line 17, in <module>
    from habitat.tasks.nav.nav import NavigationEpisode, NavigationTask
  File "/home/fei/Documents/habitat-api/habitat/tasks/nav/nav.py", line 36, in <module>
    from habitat.utils.visualizations import fog_of_war, maps
  File "/home/fei/Documents/habitat-api/habitat/utils/visualizations/__init__.py", line 7, in <module>
    from habitat.utils.visualizations import maps, utils
  File "/home/fei/Documents/habitat-api/habitat/utils/visualizations/maps.py", line 26, in <module>
    "100x100.png",
  File "/home/fei/miniconda/lib/python3.7/site-packages/imageio/core/functions.py", line 267, in imread
    return reader.get_data(0)
  File "/home/fei/miniconda/lib/python3.7/site-packages/imageio/core/format.py", line 346, in get_data
    im, meta = self._get_data(index, **kwargs)
  File "/home/fei/miniconda/lib/python3.7/site-packages/imageio/plugins/pillow.py", line 301, in _get_data
    im, info = PillowFormat.Reader._get_data(self, index)
  File "/home/fei/miniconda/lib/python3.7/site-packages/imageio/plugins/pillow.py", line 170, in _get_data
    i = self._im.tell()
  File "/home/fei/miniconda/lib/python3.7/site-packages/PIL/PngImagePlugin.py", line 827, in tell
    return self.__frame
AttributeError: 'PngImageFile' object has no attribute '_PngImageFile__frame'

hugovk, 2020-04-02:
This is a bug in the new Pillow 7.1.0 release: https://github.com/python-pillow/Pillow/issues/4509.

In the meantime, you can install `pillow!=7.1.0`

wuxx1624, 2020-04-02:
@hugovk  Thank you for the reply! However, I installed pillow!=7.1.0, but the issues doesn't solve. Do I also need to re-install any other dependency?

wuxx1624, 2020-04-02:
@ hugovk It solved! Thank you so much!

closed 2020-04-02 by wuxx1624

hugovk, 2020-04-02:
Good to hear! There's also a fixed 7.1.1 coming later today.