facebookresearch/habitat-lab #356: Controller for Habitat Agent

pushkalkatara, 2020-04-01:
Following closed issue #274 

`set_translation` and `set_rotation` is not working for the agent. Are these functions only for the objects and not for the agent? My agent id is 0 and no objects are present in the environment.

I tried this controller:
```
def controller(self, agent, velocity):
        current_state=agent.get_state()
        print("OLD state")
        print(current_state)
        self._sim.set_translation(np.array([0.0, 0.0, 0.0]), 0)
        Q = quat_from_angle_axis(np.pi, np.array([0.523599, 0.523599, 0.523599]))
        self._sim.set_rotation(quat_to_magnum(Q), 0)
        current_state=agent.get_state()
        print("New state")
        print(current_state)
```
Agent state
```
OLD state
AgentState(position=array([-1.,  0.,  8.], dtype=float32), rotation=quaternion(1, 0, 0, 0), velocity=array([0., 0., 0.]), angular_velocity=array([0., 0., 0.]), force=array([0., 0., 0.]), torque=array([0., 0., 0.]), sensor_states={'color_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'depth_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0))})
New state
AgentState(position=array([-1.,  0.,  8.], dtype=float32), rotation=quaternion(1, 0, 0, 0), velocity=array([0., 0., 0.]), angular_velocity=array([0., 0., 0.]), force=array([0., 0., 0.]), torque=array([0., 0., 0.]), sensor_states={'color_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'depth_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0))})
```
 

I also tried with the following controller:
```
def agent_controller(self, agent, velocity):
        #_X_AXIS = 0
        #_Y_AXIS = 1
        #_Z_AXIS = 2
        print(velocity)
        axises = [0,1,2]
        current_state=agent.get_state()
        print("OLD state")
        print(current_state)
        for axis in axises:
            # Translate
            ax = agent.scene_node.transformation[axis].xyz
            agent.scene_node.translate_local(ax * velocity[axis])
            # Rotate
            _rotate_local_fns = [
                SceneNode.rotate_x_local,
                SceneNode.rotate_y_local,
                SceneNode.rotate_z_local,
            ]
            #SceneNode.rotate_x_local(agent.scene_node, mn.Deg(velocity[axis)
            _rotate_local_fns[axis](agent.scene_node, mn.Deg(velocity[axis+2]))
            agent.scene_node.rotation = agent.scene_node.rotation.normalized()
        
        current_state=agent.get_state()
        print("OLD state")
        print(current_state)
```

the output was:
```
OLD state
AgentState(position=array([-1.,  0.,  8.], dtype=float32), rotation=quaternion(1, 0, 0, 0), velocity=array([0., 0., 0.]), angular_velocity=array([0., 0., 0.]), force=array([0., 0., 0.]), torque=array([0., 0., 0.]), sensor_states={'color_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'depth_sensor': SixDOFPose(position=array([-1. ,  1.5,  8. ], dtype=float32), rotation=quaternion(1, 0, 0, 0))})
NEW state
AgentState(position=array([-1.,  0.,  8.], dtype=float32), rotation=quaternion(0.933012723922729, 0.0669873058795929, 0.250000029802322, 0.25), velocity=array([0., 0., 0.]), angular_velocity=array([0., 0., 0.]), force=array([0., 0., 0.]), torque=array([0., 0., 0.]), sensor_states={'color_sensor': SixDOFPose(position=array([-1.649519,  1.299038,  8.375   ], dtype=float32), rotation=quaternion(0.933012723922729, 0.0669873058795929, 0.250000029802322, 0.25)), 'depth_sensor': SixDOFPose(position=array([-1.649519,  1.299038,  8.375   ], dtype=float32), rotation=quaternion(0.933012723922729, 0.0669873058795929, 0.250000029802322, 0.25))})
```
with this controller agent is rotating but the magnitude is incorrect. I found out this by looking at the quaternion. The quaternion after the controller was 
```
rotation=quaternion(0.933012723922729, 0.0669873058795929, 0.250000029802322, 0.25)
```

but if I consider the ground truth quaternion after rotation of agent by 30 degrees (0.523599 radians),
it should be
```
[ 0.2529195, 0.2529195, 0.2529195, 0.8989412 ]
```

How should I design the controller?

mathfac, 2020-04-02:
cc @aclegg3 on the controller for the agent.

aclegg3, 2020-04-03:
Hey @pushkalkatara, thanks for the question!

Yes, these functions will only work on objects. That said, sounds like you might be interested in our recent velocity control implementation:
 - [Attaching agents to the object/bodies](https://github.com/facebookresearch/habitat-sim/pull/492#issue-374969319) this can be done in python currently and will enable `set_rotation` and `set_translation` to modify the agent transforms. Just pass your agent scene node into `add_object` to attach it.

 - [Constant velocity control of objects](https://github.com/facebookresearch/habitat-sim/pull/486#issue-373416399) I realize there are no python bindings for this yet. I'm happy to get those setup. There will also need to be some changes to the step filter to use this with the NavMesh for navigation constraints if you want to use that feature.

FYI, these features are fresh and better docs are coming soon.

pushkalkatara, 2020-04-04:
Hi @aclegg3 

Thanks for pointing me to the links. I am currently using `set_rotation` and `set_translation` to modify the agent transforms in the `controller` function above, but the `new_state` fetched after applying the quaternion is not the same as desired(ideal scenario).

I tried extending the python bindings with `velocity_control` functions. Here is the PR for the same - https://github.com/facebookresearch/habitat-sim/pull/565 . I am unable to access the functions in python after rebuilding (added test in the PR which is failing). It would be helpful if you could point out what am i missing. 

Thanks.

rpartsey, 2022-09-04:
Relates to https://github.com/facebookresearch/habitat-sim/pull/565, solved by https://github.com/facebookresearch/habitat-sim/pull/571.

closed 2022-09-04 by rpartsey