facebookresearch/habitat-lab #362: Habitat API changes episode goal height

smorad, 2020-04-06:
## 🐛 Bug
After a few epochs, the y value of goals in the simulator seems to diverge from the y value of goals in the dataset. This is validated by comparing `episode{episode_id=1234...}` from `train.json.gz` to `self.habitat_env.current_goals[0].position`. I've checked four specific episodes, and the Y offset seems to be exactly `0.55` or `0.5` in those cases. Agent starting position is not affected.

This is an issue, as calls to `geo_dist` will return `inf`, as in the path is not navigable. For reference, my agent height is 0.18m, so I'm not entirely sure where this difference is coming from.

## Command

## To Reproduce
I am currently reproducing this in a complex, non-portable environment. All I do in the simulator is load a config, then call step/reset. Reproduction forthcoming.

## Expected behavior
I'd expect the goal y values in the episodes on disk and in the simulator to match.

## Additional context

smorad, 2020-04-06:
It seems there is an open issue in `habitat-sim` that discusses this: https://github.com/facebookresearch/habitat-sim/issues/405

This only occurs after running a good amount of episodes (>200), could this be some sort of race condition?

smorad, 2020-04-06:
Closing and moving to the `habitat-sim` repo.

erikwijmans, 2020-04-06:
Hi, that issue you linked to in habitat-sim seems unrelated as that is about folks wanting to be able to teleport the agent around in the environment (which if done unconstrained will lead to the agent entering a different connected component of the environment).

Can you tell me what `self.habitat_env.current_goals` is?  That isn't something in master.

smorad, 2020-04-06:
Sure, this is from the dataset
```
{"episode_id": "41558", "scene_id": "/root/data/habitat/gibson/Cooperstown.glb", "start_position": [1.4900641441345215, 0.18463699519634247, 2.322016954421997], "start_rotation": [0, 0.9258610711618118, 0, -0.3778640984627972], "info": {"geodesic_distance": 5.514394283294678, "difficulty": "easy"}, "goals": [{"position": [2.135885238647461, 0.18463699519634247, 6.5982666015625], "radius": null}], "start_room": null, "shortest_paths": null},
```
This is `current_goals[0]`:
```
NavigationEpisode(episode_id='41558', scene_id='/root/data/habitat/gibson/Cooperstown.glb', start_position=[1.4900641441345215, 0.18463699519634247, 2.322016954421997], start_rotation=[0, 0.9258610711618118, 0, -0.3778640984627972], info={'geodesic_distance': 5.514394283294678, 'difficulty': 'easy'}, goals=[NavigationGoal(position=[2.135885238647461, 0.7346369951963425, 6.5982666015625], radius=None)], start_room=None, shortest_paths=None)
```

erikwijmans, 2020-04-06:
How is it set?  

I haven't seen this personally and don't have any idea how to reproduce currently.  There is nothing in habitat-api or habitat-sim that would cause that field to get modified as far as I know.

smorad, 2020-04-06:
Bug on my end, apologies for opening this issue. I was spawning objects near the goal, and left out a call to `copy()`, accidentally incrementing the actual goal height ever so slightly for each object spawn. The error only became visible after cycling through all the episodes many times.

Errors never become apparent until you make a fool of yourself :P

closed 2020-04-06 by smorad

erikwijmans, 2020-04-06:
It happens to everyone :) glad you found the bug!