facebookresearch/habitat-lab #366: Growing CPU usage during ObjectNav baseline training

obin-hero, 2020-04-08:
## ❓ Questions and Help
Hi. I'm currently training ObjectNav task baseline using ddppo.
I just edited the config file to increase the number of processes.
Additionally, I added episode filter in make_dataset func so that only easy episodes (geodesic distance < 3m) can be included in the dataset. 
Except for these modifications, I nearly touched nothing.

My setting for the number of processes is **4.**
As training proceeds, cpu memory usages for each processes just keep growing.
When I first started the code, the whole processes only take 10~12GB.(approximately)
After training for 3 hours, (336.4k frames) they are occupying almost **40GB** + alpha and it doesn't seem to converge.

I debuged the cpu usage using pustil library, but pytorch network part or rl training part don't have any problems.
I think the simulator environment has problem in the memory management. Because it seems that when the scene changes in certain process (deconstructing the old scene and constructing the new scene) , the memory of the process shows increase. However, I'm not so certain about this.

Is it the right phenomenon when training with multiple processes? 
Can you tell me how much is the ideal RAM size for multi-process training?
Is there anything I can do to limit the cpu usage?

erikwijmans, 2020-04-08:
~40GB for 4 processes sounds right once the objectnav dataset is fully loaded into memory (i.e. check memory usage after about 5 minutes and it should be ~40GB).

On current master of habitat-api, memory usage will continue to grow very very slowly through the course of training due to some caching -- this is relatively light tho, even after processing 100k episodes, this won't use more than ~4GB across all workers. With a little change to habitat-api, this can be made to not grow however -- adding
```
if self._current_episode is not None:
    self._current_episode._shortest_path_cache = None
```
here: https://github.com/facebookresearch/habitat-api/blob/master/habitat/core/env.py#L199
With this change, memory usage is perfectly stable when scenes aren't switching.  When a scene switch happens, memory usage will necessarily change, but nothing appears to be leaking.

obin-hero, 2020-04-12:
Thank you for your reply 😀
The memory usage became perfectly stable with adding the code lines you suggested. (Now training has been going on for almost two days with stable memory status)

It seems that the ObjectNav task  + habitat simulator itself needs a lot of memory. 
I am using a shared server with my colleagues, so I was worried about my code will occupy all the memory and eventually shut down all the other running codes in the server.

Thanks for your comment again, 
Have a nice day!

closed 2020-04-12 by obin-hero