facebookresearch/habitat-lab #373: How to acquire the shortest path for Objectnav task

qweas120, 2020-04-16:
## ❓ Questions and Help
I try to replace the config file in [https://github.com/facebookresearch/habitat-api/blob/6091d0aedd41d348824252d74119d5d0d3355b8e/examples/shortest_path_follower_example.py#L65](/examples/shortest_path_follower_example.py Line65) with `configs/tasks/objectnav_mp3d.yaml`, but it seems that the agent always gets stuck in some corners. Here are some visualized results I uploaded to [https://drive.google.com/drive/folders/1H50pEvO_sQRYQzp0CZ9guj0pbXMBt3me?usp=sharing](GoogleDrive) It might due to that the sliding is not allowed. How to get the shortest path for Objectnav task?
BTW, I find a keyword 'shortest_path' in the training set `xxxxxxxx.json.gz`. What's the difference between this _shortest path_ and the shortest path in the example. Thanks
```
{
            "episode_id": "0",
            "scene_id": "mp3d/17DRP5sb8fy/17DRP5sb8fy.glb",
            "start_position": [
                -1.89078,
                0.07245,
                -2.1819
            ],
            "start_rotation": [
                0,
                0.62436,
                0,
                -0.78113
            ],
            "info": {
                "geodesic_distance": 5.56265,
                "euclidean_distance": 5.22924,
                "closest_goal_object_id": 5,
                "navigation_bounds": [
                    [
                        -11.59344,
                        -0.12755,
                        -5.39202
                    ],
                    [
                        4.75703,
                        4.27878,
                        2.88662
                    ]
                ],
                "best_viewpoint_position": [
                    -6.98474,
                    0.07245,
                    -1.70629
                ]
            },
            "goals": [],
            "start_room": null,
            "shortest_paths": [
                [
                    3,
                    3,
                    3,
                    1,
                    3,
                    1,
                    3,
                    3,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    3,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    1,
                    null
                ]
            ],
            "object_category": "bed"
        },
```

mathfac, 2020-04-16:
@qweas120, please re-download ObjectNav dataset and update your habitat-sim/habitat-api to latest master or `challenge-2020` tags and re-try.

qweas120, 2020-04-17:
Actually I try the examples using the provided docker image. Do I need to rebuild habitat-sim/habitat-api inside the container? Or I only need to pull the latest docker image?

qweas120, 2020-04-17:
I download the latest master branch and use fairembodied/habitat-challenge:testing_2020_habitat_base_docker docker image to run the example. The problem still not solved.

closed 2021-05-26 by HanqingWangAI