facebookresearch/habitat-lab #377: introduce sensor noise

rosanom, 2020-04-21:
Hi! I would like to introduce a noise in the sensor readings (gps, compass, etc). Is the `habitat/tasks/nav/nav.py` the correct file I can work with? I've seen I could act on the `get_observation` method of every single sensor to "disturb" their output. 
My concern is about makes it work with the ddppo baselines, because I've already tested it successfully with the `test/test_sensors.py` but when I run the ddppo baseline it seems the `nav.py` file is never called.
Thanks in advance!

erikwijmans, 2020-04-21:
By "`nav.py` file is never called" I assume you mean the sensor you made isn't showing up in the registry?  Or the changed sensors aren't?  Make sure your changes are being picked up correctly, i.e if you didn't re-install after editing, make sure to add `/path/to/habitat-api` to your PYTHONPATH.

rosanom, 2020-05-11:
The problem was that all the changes were ignored during train, even after re-installing the library. I had another local copy of the repo and when I was launching the baseline sh script, the script of that repo were executed. Update my PYTHONPATH has solved the problem. Thanks!

closed 2020-05-11 by rosanom