facebookresearch/habitat-lab #381: Changing the height of image sensor

qianLyu, 2020-04-27:
## ❓ Questions and Help
Hi all,

Currently I hope to change the height of the image sensor while training habitat-api RL baseline. I'm trying to use the function 'set_transformation_from_spec()' to change the sensor height. However, this function only changes the height of the agent(default 1.5m), not the height of the sensor(default 1.25m). Thus, the sensor height remains unchanged. I'm wondering, if there is issue in the function 'set_transformation_from_spec()'? Or, if there are other ways to easily change the sensor height in the training process?

Thank you!

erikwijmans, 2020-05-01:
The functionality to do this isn't exposed too well, but it does work in my testing, i.e. doing:

```
agent = sim.get_agent(0)
agent.agent_config.sensor_specifications[0].position = [0, 1.5, 0]
for _, v in agent._sensors.items():
    v.set_transformation_from_spec()
```

successfully moves the camera up to 1.5m above the ground from the initial 1m I placed it at.

qianLyu, 2020-05-05:
Thanks Erik! Your code also works well in my test environment! I made a tiny mistake in the configuration process.

closed 2020-05-05 by qianLyu